F-Army/arduino-dw1000-ng #89: Add the possibility to not use interrupt but to poll for them

an4cr0n, 2018-09-27:
This will decrease performance but make the program more stable because there will never be interrupt when they (the interrupts) are locked.

Elon-Wang, 2018-10-10:
Hello~
I come from the project of Thomas' project and work on this modules for a while, and there is a problem confused me for a long time. 

Although I read most of Thomas' code, including DW1000Raing.cpp and DW1000.cpp. I don't really understand when will the interrupt happens. 

When I source back to the attachInterupt() function, I know it happens when the IRQ pin changes but I don't know when the IRQ pin change. I guess it will be changed when it sent/receive a message, but I don't see any codes about changing that pin, is it depends on the inside circuit or I not found it?

Thanks for your reply!!!

an4cr0n, 2018-10-10:
Yeah it is an inside mechanism of the dw1000 itself, there is a register
inside the chip that controls when to send an interrupt relative to the
selected elements.(we add a struct and a method to select them)

Elon-Wang, 2018-10-10:
> Yeah it is an inside mechanism of the dw1000 itself, there is a register inside the chip that controls when to send an interrupt relative to the selected elements.(we add a struct and a method to select them)
> […](#)

Thank you for your reply, that really helps me a lot!! I finally find someone who can answer my question.😊😊

After I set the antenna delay, the one to one ranging become precise however when I use three anchors and one tag to do the ranging, there are only one pair of ranging turn out to be reliable, while the other two are quite wrong. The anchor with right raging data is random. For example, if I have three anchors A B C, and turn them on, only A get the right ranging data. However, If you turn anchor A off, one of the rest anchor will work well, the other is still reading the wrong distance. The same phenomenon happens when you turn the second anchor off.

Then I turn them down and using the anchor which is works well to print out the timestamp of Poll/PollACK/Ranging , and calculate the reply1 round1, reply2 , round2. It's strange to see reply1 is bigger than round1 sometime, which is ridiculous. However, those ridiculous timestamps can calculate out the right distance using the formula (Tround1 * Tround2 - Treply1 * Treply2)/ (Tround1 + Tround2 + Treply1 +Treply2) .

Have you ever encounter with the same problem? Is this out of the non-calibration of crystal oscillator,if yes,how to do it? If you know the reason, please tell me how to solve it.

I'm very grateful!!!

Elon-Wang, 2018-10-10:
So, you haven't try multiple anchors locate yet?
We can discuss later, lol
Time is very late in my time zone and I have lots of courses in tomorrow~
I'll open an issue soon and introduce more about my issue.
Anyway, thanks a lot for your help.

an4cr0n, 2018-10-10:
Yeah, we rewrote a lot of the thotro's driver, this is still a work in progress, we wanted to get rid of DW1000Ranging as a loop and instead to make it a library compliant with ISO/IEC-24730/62 to leave the user with more control on the action flow.
I was asking about how you did the 3 anchor range because thotro's library doesn't use frame filtering, so every anchor will be receiving the messages from the tag and probably overlapping when responding. When we finish making the examples using the ISO standard MAC format we will check if the frame filtering code. For the moment you could try to add a destination address inside the sent data and check at the receiver if it was for it and discard other anchor's frames.(the standard uses both source and destination addresses).
Feel free to ask any questions!

Sonic0, 2019-01-11:
I think that we could close this issue @an4cr0n

Sonic0, 2019-02-20:
done

closed 2019-02-20 by Sonic0