F-Army/arduino-dw1000-ng #111: Wrong distance read with three anchor working together

Elon-Wang, 2018-10-11:
Hello~ 
I am doing a university project, and using thoros' code. Thoro's project is not active and stop to develop, so I always find nowhere to communicate and solve problems, it confused me for a long time until I found you.(maybe because i'm not familiar with github)

As for my part of DW1000 research. I simple use thotro's code, and do a little modification to meet my own demand. I read and understand most source code in his project like DW1000.cpp, DW1000Ranging.cpp etc. At first, the ranging always have a drift in distance, after I change the antenna delay, the drift disappear. 

However, my goal is to use three anchors to locate the tags. So I open the three anchors at same time. But there are only one anchor reads the right data, while the other two anchors read very wrong data. and it's strange that, when I only turn off the well-working anchor, one of the rest anchor will reads the right distance. Same things happen when I turn off the second well-working anchor, and the only left anchor will reads the right distance. This is first of my problem.

Then I use some debug code trying to find out why this strange things happen, I print out the time stamp to get the Treply and Tround. However, I found that Treply is bigger than Tround sometime(not seldom but frequently appear in a bulk of continues ranging data). We know in the ranging algorithm, Tround = Treply +2 * Tof (time of flight). However, those ridiculous can calculate out the right distance (after antenna delay). This is my second problem.

And I have briefly read your example TwoWayRangingInitiator/Responder, and I think maybe you haven't start to develop the multiple anchors and multiple tags system. And I test your code in my device and found your ranging accuracy is very high, plus no timestamp error as I found in thoro's project. If you want to develop the multiple system, I'd like to help. 

Here are my calibration system and my device: 4 * DWM1000; 4 * Arduino UNO R3

[image]
[image]

And this is data when I test your example TwoWayRangingInitiator/Responder, the ripple is around ±3 cm, which is every good, very few point got extreme big distance having 15cm error can be removed by filter.
[image]
[image]

an4cr0n, 2018-10-12:
I'm glad the ranging is very accurate, having feedback from other sources helps a lot to make sure the changes are working for everyone. We'll start working on the 3 anchor position when we finish to make the ranging messages on the example RTLS ISO standard compliant (almost done). Using the ISO way of doing two way ranging should not give you those errors in the distance as you can check destination and source (just like TCP/IP MAC Layer), that's because i think that thotro's code erroneously makes the 3 ranges of the anchors in parallel so the timestamps are only right on one.
I'm not sure of this because we didn't like how the DW1000Ranging was implemented as a loop so we started to work on our own fork. Hope to publish the new ranging fast so we can all test it out.
Thank you, as always feel free to open issues and help with the project, we'd like to expand our community.

Elon-Wang, 2018-10-12:
Thank you for your reply!!!
Actually, making 3 ranges of the anchors in parallel is called broadcast in communication, and I don't think this part will go wrong. Although it making 3 ranges of anchors in parallel, it's also have the part of checking destination and source. The frame broadcast to every anchor but the data inside it have the destination check( different anchors with different address and corresponding to different wait time). 
And the official datasheet( user manual ) also give a picture to describe the theory of multiple tags & anchors system.(shown as below)
[image]

Everything of thoro's code looks non-error from my perspective however, turning out to calculate the wrong distance.😭 It's really confused me. And I think maybe we can have more communicate to help each other to make the targets~ 😃

an4cr0n, 2018-10-12:
Oh yeah i forgot about this way of multiple ranging as in the standard you cannot do that as there is no concept of 'anchor' but rather the concept of infrastructure (the anchors are like the communication channel of this system and two way ranging is strictly a flow of messages between tag and the infrastructure)

You can contact me on my email (it's in the github profile) to exchange an other way to communicate faster. Thank you for your support.

Leylan24, 2019-07-16:
Hi, im planning on using 3 anchors as well have you found a solution?

Sonic0, 2019-07-16:
> Hi, im planning on using 3 anchors as well have you found a solution?

Hi Leylan24, 3 anchors are supported but not in parallel. The radio protocol does not support this feature in the same frequency channel.

maeenk, 2020-10-31:
Hello Again,

I decided to switch from using the NodeMCU to an Arduino Uno. I have been having a hard time with the hardware parts of things. I have finally been able to get the basic sender and receiver examples to run. Finally seeing "Hello" on the receiver side. 

I went ahead and started testing for two way ranging. However, these are the outputs I am seeing. 

Two Way Responder
[image]

Two Way Initiator
[image]

I had added a pull up resistor on the IRQ as mentioned from a "closed issue", but the above screenshots are what I see. I see the same result even without the pull up resistor on the IRQ.

Any suggestions?

Sonic0, 2020-10-31:
Hi @maeenk , You have setted up two different types of configuration for radio communications (last line, Device Mode). In addition, IRQ requires an pull-down resistor to ground.

maeenk, 2020-11-04:
The two different types of configuration. Does this mean that two different files were used? So would re-downloading the zip file fix this issue?