F-Army/arduino-dw1000-ng #128: The main anchor can't receive message from anchor C.

Elon-Wang, 2018-10-24:
To begin with, I use the Arduino UNO R3. and use 5V to supply the DW1000 chips.

The situation is that, main anchor can range with the tag, and so does the anchor B, so main anchor can receive the ranging report from anchor B.

However, main anchor can't receive message from anchor C, even I tried to use different DW1000 chips to have the test, the main anchor still can't find anchor C. So, this might not be a hardware problem.

Elon-Wang, 2018-10-24:
Try to print out the message on tag, and found it only receive the message from anchor B for once, seem like it doesn't receive the ranging confirm message. But the main anchor can receive the ranging report message, and anchor B can print out the ranging result.

[image]

Elon-Wang, 2018-10-24:
The problem can be solved by add a delay between the transmitActivityFinished() and transmitRangeReport() these two method like this,
[image]

If you sending the message from MCU to DW1000 chip too frequently, this might cause a problem.
When the chip is deal with the old message when you send in a new message to the chip, it will lost the first message and only send out the second messages, so main anchor can receive ranging report but tag can't receive ranging confirm.

Sonic0, 2018-10-28:
I think that this Issue should be closed. Fixed with #127

Elon-Wang, 2018-10-28:
Yeah, thank you for reminding。

closed 2018-10-28 by Elon-Wang