F-Army/arduino-dw1000-ng #139: Example TwoWayRanging is not work. (This may be hardware problem)

tdh8316, 2019-01-24:
### Step 1: What do you want?
* TwoWayRangingInitiator and TwoWayRangingResponder do not work. therefore I want them to work.

### Step 2: Describe your environment
 * Arduino IDE version or Toolchain: Arduino IDE 1.8.7
 * Arduino device: Arduino Uno R3
 * Used adapter board: None
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * TwoWayRangingResponder.ino
  * TwoWayRangingInitiator.ino

#### Observed Results:
[image]
It seems not to work!

Thus I tried Basic Connectivity Test.
[image]
I'm not sure it's correct output because I'm a beginner of Arduino and DWM1000.
But I don't think there is any noticeable problem.

And then I uploaded Basic Sender/Receiver to Arduino.
[image]

I still have no clue about this, but it seems to work fine.

#### Main question
Why are BasicSender and Receiver working but TwoWayRanging not working?

#### Expected Results:

  * Jaja, I expected they work.

tdh8316, 2019-01-24:
Add a wiring diagram

tdh8316, 2019-01-25:
This turns out to be a hardware problem.

closed 2019-01-25 by tdh8316

tdh8316, 2019-01-26:
Completely resolved after adding a 10K pull-up resistor according to [this datasheet](https://www.decawave.com/wp-content/uploads/2018/09/dwm1000-datasheet-1.pdf).

Leylan24, 2019-09-04:
Where did you put the 10K resistor ? I am having the same issue my Basic Receiver and Basic Sender is working but the Two Way Ranging example doesnt work

tdh8316, 2019-09-06:
@lgr9603

Sonic0, 2019-09-06:
> Where did you put the 10K resistor ? I am having the same issue my Basic Receiver and Basic Sender is working but the Two Way Ranging example doesnt work

I think that the @tdh8316 's solution is relative to section 5.2.2 inside DWM1000 datasheet

lgr9603, 2019-09-08:
@Leylan24 I'm a coworker of @tdh8316 .
Well, we put the 10K resistor on pin 2 to remove noises flowing through it.
And drop the voltage of digital pins from 5V to 3.3V. Dwm1000 can work between 2.4V and 3.75V. (Along to datasheet on decawaver.com)
As I said, arduino uno digital pins use 5V, bout DWM1000 sends output signals in 3.3V. So we should increase voltage of DWM1000 signal from 3.3V to 5V. 
[image]

Leylan24, 2019-09-08:
Thanks for you reply. I'm using the ESP8266 Nodemcu and get very inaccurate ranges no idea why

Leylan24, 2019-09-22:
This is a bit of a stupid question.

But does the ranges only print when the IRQ pin gets pulled high?

Sonic0, 2019-09-23:
> This is a bit of a stupid question.
> 
> But does the ranges only print when the IRQ pin gets pulled high?

With this library, the IRQ pin is not required. There are two different init processes, with or without Interrupt pin.