F-Army/arduino-dw1000-ng #140: Sending a Rang to Tag

NimaaZx, 2019-02-25:
Hello.
I want to improve your library for my goal. I'm searching about a way to implement sending Rang to Tag side.
in this method A tag side can print its Rang from each anchor that it blinks with.
right now we don't have it.

We read dwm1000 manuall.If i understand correclty there are some optional ways to do that.
I understood the tag and anchor connect in this way :

Tag : blink
Anchor : Ranging_INIT
Tag : Poll
Anchor: Poll-Ack
Tag: Sending Final Message to anchor
Anchor: have Rang

at this time Anchor calculates and have a Correct Rang then print it but Tag does not have. it does not print anything right now.

in the best solution We should send a Rang String to Tag after "Final Message" from anchor.

I found some lib codes in each side, tag and anchor.step by step what happend.
I could send a packet frame with "0x60" value from anchor side,
but I could not find a way to wait to recieve this packet frame "0x60" in Tag side.

I don't know what "command - code - function"  i should use to wait in Tag side to receive "0x60" packet frame in correct time.

I mean the waiting time for receiving packet in Tag side.(halt time)

Sonic0, 2019-03-04:
Hi NimaaZx,
according to the standard ISO/IEC 24730-62_2013 on which we are based, the final step of the TWR  flow (with embedded TX time) is an Activity control (0x10 , 1 octet) + Activity Code ( 0x00 or 0x01 , 1 octet) + Optional parameters ( new blink rate or Next anchor to perform ranging ,2 octet ), that the anchor sends back to the tag.
According to the standard, the final step isn't usable to report range to the tag.

You Should create a new personal message (and functions) with User Code (from 0x60 to 0x77 and from 0xE0 to 0xF7) that returns the range to the tag. Then the Tag should implement a function to receive the range.
In this scenario, you can't use high-level abstraction functions inside DW1000NgRTLS but you must use the individual functions for every single step of the TWR flow ( transmitTwrShortBlink(), transmitRangingInitiation(), transmitPoll(), and so on ... ) + new personal function.

We haven't implemented this functionality because it isn't a default step in the standard 24730-62

NimaaZx, 2019-03-06:
thank you for your response.
We implemented it inside "TransmitRangingConfirm"
there is a packet that we put the range in and when this function Call, it is sent and in tag side we receive it.

I have 3 more questions .

1- For developing and working with dwm1000 module,
what do we do if someone like you don't share his libs and source code with us? you write all libs ? how did you do that? how much time does it take ?
I mean if your source code does not exist what should we do for working with this module?
because I never found the library from decawave and we want to build a industrial product with dwm1000 and esp32.
we need them.

2- we use your souce code and library with some our modifications.
do you set sleep mode for anchors?
because after 10 minutes when a tag is off or out of coverage , when the tag is back , anchors can't see them until we reset anchors manually.
 
3- do you have any TX/RX mode for anchors or tags coverage or something like said in 
TX/RX table in datasheet ?
I see something like AutoTx in your code.

thanks a lot.

Sonic0, 2019-03-06:
You are welcome!

1) Before all, you must read DW1000 user manual and DWM1000 datasheet. It's very important to understand how to develop a driver and libraries for the DWM1000. We spent approximately 1 month only to study these two documents, plus 3 months (more or less) to refactor the Thotro code and add some futures. Our code is compatible with esp8266 but with esp32 we had problems. 
The best work that you can do is: take the source code of the driver from Decawave and convert it for the esp32 and esp-idf (driver is written in C).

2) No. Sleep mode isn't enabled in anchors. Maybe that is a bug in our code or in your modification of the TransmitRangingConfirm function. Remember that the last 2 octets (payload is: 0x10+0x01+NextAnchor) are for specifying the next-anchor to perform ranging.

3) before reply to this question, I prefer consulting with my friend @an4cr0n

NimaaZx, 2019-03-11:
Thank you for your concern.
1- Actually at first We did not find a source code of dwm1000 driver in official site. (can you send me a link please ?) 
as well as i think reading module datasheet is a little hard for us. because I think 
Electrical Engineering or other knowledges is needed. but we did it.
we studied software engineering. some topics are related to our majors. even writing library with low level language is a challenge.
I mean this indoor positioning is a part of of IoT. 
unfortunately IoT is a branch that requires several fields of study(software-hardware-electronic eng)

2- at this time I dont have an access to our code but as I tell you we injected ranging parameter to "TransmitRangingConfirm" function. yes you right, we injected it after "next anchor bits". I mean after last bit you wrote, we injected ranging parameter and sent to the packet to tag side.
I will send the code for you.
for sleep problem we used reset command.
it solves the problem for now.

3- About Tx,Rx / for Extending Rang of dwm1000.

I found this issue of Thotro topic
https://github.com/thotro/arduino-dw1000/issues/245

Do you use this source for extending rang of dwm1000 ?

somebody said using this 
writeValueToBytes(txpower, 0x1F1F1F1FL, LEN_TX_POWER);

Sonic0, 2019-03-11:
1) Drivers are inside [_DW1000 Application Programming Interface software package_](https://www.decawave.com/wp-content/uploads/2018/09/dw1000_api_rev2p04-1.zip).
Yes, IoT is a branch that requires several fields of study.

2) OK

3) The range of the dwm1000 is very short because it has a very small antenna. You can try to extend it changing frequency(channel, lower is better), use manual power instead of SmartPower (see the struct of the configuration options). This argument is explained in DW1000's user manual or in an Application note.
Very important is the vertical antenna polarization for the DWM1000 module.

closed 2019-05-27 by Sonic0