F-Army/arduino-dw1000-ng #141: About Multiple Tag Support

hknckr, 2019-05-17:
Hi, I am trying to work with multiple tags.  I mean  at least 1 anchor and 2 tags at the same time
(thotro's library don't have this support yet, If you put second tag in to the test environment everything goes bad)

I couldn't find any info about multiple tag support in your wiki. 

Is your library has multiple tag support? 

Thanks

Sonic0, 2019-05-19:
Yes, Tweaking the examples ( mainly the address section ) you should obtain multiple tag support.
What do you mean one anchor at least? At least 3 anchors are needed to localize.

hknckr, 2019-05-19:
I know at least 3 anchors needed to localize. I have got 5 DWM1000 right now but only assembled 3 of them, because of that I was trying one of them as anchor and the others are as tag. Thank you for your answer.

closed 2019-05-19 by hknckr

Sonic0, 2019-05-19:
With 3 DWM1000 you can only perform Raginging with 1 Anchor and 2 Tags.
When you will have 3 Anchors and 2 Tags the localization is possible setting the right time slot through DWM sleep and delays, this isn't implemented with high level function in our library.
There are some White Papers on this problem. Moreover there are the Application Notes from Decawave ( ex. APS013 ).