F-Army/arduino-dw1000-ng #142: Sending and Receiving From Multiple Channels

hknckr, 2019-05-23:
Hi,

I am trying to achieve TDOA approach and biggest problem here is to sync clocks. I am trying wireless clock sync. 

The main idea here is to use 4 anchors, 1 is going to be the main anchor and send timestamp sync packages to other 3 slave anchors. 

And TAGs are going to send blink messages to those 3 anchors and anchors will caculate TDOA's.

And I want to seperate sync and tag blink requests by sending and receiving them from different channels to use bandwidth more effieciently.

With these scenario I have some questions.

1 - Is it possible to achieve this ? Because 3 Slave Anchors needs to listen 2 channels at the same time? (Both tag blinks and Main Anchor sync packages) 

2 - Is it possible to achieve wireless sync with such method? Because 1 nanosecond error in calculations means 30cm error localization.

There are companies achieved wireless sync with such method but it is not clear how they achieved this. May be they purchased IP from Decawave about Wireless Sync implementation.

I just want to here from you, and your ideas.

Thank you...

Regards

Sonic0, 2019-05-23:
Hi hknckr,
Transfer or receive at the same time in the same channel is not possible and change channel each time is a waste of time.

To this question, there are a lot of Papers. I can't publish the URL of these, but some titles are:
- High-accuracy TDOA-based Localization without Time Synchronization
- Comparison of wireless clock synchronization algorithms for indoor location systems
- A new method for wireless synchronization and TDOA error reduction in UWB positioning system
- Synchronization-free and Low Power TDOA for Radio Based Indoor Positioning ( This is WOW!! )

To search more White Paper use Google Scholar.

Notice, the Tag payload, to perform TDoA, is different from ToF. If you would like to respect the standard  ISO/IEC 24730-62 ( as in our payloads ), you should buy this standard.

If you will write a code to extend this driver and libraries you can make a pull request, maybe for someone your work is useful.

I hope to have been helpful

hknckr, 2019-05-23:
Thank you for your answer, I am trying to do my best. My hardware information is not enough about DWM1000 yet so it is going a little bit slow but I will let you know if I have any progress. 

By the way I will be checking your recommendations about TDOA. 

Thank you again...

I am closing this issue , I will open new one if needed again

closed 2019-05-23 by hknckr