F-Army/arduino-dw1000-ng #144: Arduino, ESP32, fails to compile

zrecore, 2019-06-01:
### Step 1: What do you want?

* The library examples do not compile when ESP32 boards are selected.

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: Arduino IDE 1.8.9, with ESP32 Arduino board package
 * dw1000ng Library version: Whatever version this is.
 * Arduino device: ESP32-PICO-D4, DWM1000
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * BasicConnectivityTest example

#### Steps to reproduce (for bugs only):

  1. Open the BasicConnectivityTest example
  2. Select "ESP32 PICO Kit" as the board
  3. Attempt to compile

#### Observed Results:

  * What happened?  This could be a description, log output, etc.
```
Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp: In function 'RangeRequestResult DW1000NgRTLS::tagRangeRequest()':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:175:51: error: narrowing conversion of 'DW1000NgUtils::bytesAsValue((& init_recv[13]), 2)' from 'uint64_t {aka long long unsigned int}' to 'uint16_t {aka short unsigned int}' inside { } [-Werror=narrowing]
         return { true, DW1000NgUtils::bytesAsValue(&init_recv[13], 2) };
                                                   ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp: In function 'RangeResult DW1000NgRTLS::tagFinishRange(uint16_t, uint16_t)':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:208:64: error: narrowing conversion of 'DW1000NgUtils::bytesAsValue((& act_recv[11]), 2)' from 'uint64_t {aka long long unsigned int}' to 'uint16_t {aka short unsigned int}' inside { } [-Werror=narrowing]
                 return {true, true, DW1000NgUtils::bytesAsValue(&act_recv[11], 2), 0};
                                                                ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp: In function 'RangeInfrastructureResult DW1000NgRTLS::tagRangeInfrastructure(uint16_t, uint16_t)':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:231:35: error: narrowing conversion of 'result.RangeResult::new_blink_rate' from 'uint32_t {aka unsigned int}' to 'uint16_t {aka short unsigned int}' inside { } [-Werror=narrowing]
             return { true, result.new_blink_rate };
                                   ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRanging.cpp: In function 'double DW1000NgRanging::correctRange(double)':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRanging.cpp:81:5: error: control reaches end of non-void function [-Werror=return-type]
     }
     ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:236:5: error: control reaches end of non-void function [-Werror=return-type]
     }
     ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp: In function 'RangeResult DW1000NgRTLS::tagFinishRange(uint16_t, uint16_t)':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:215:5: error: control reaches end of non-void function [-Werror=return-type]
     }
     ^
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp: In function 'RangeAcceptResult DW1000NgRTLS::anchorRangeAccept(NextActivity, uint16_t)':
/Users/aalbino/Documents/Arduino/libraries/arduino-dw1000-ng/src/DW1000NgRTLS.cpp:301:5: error: control reaches end of non-void function [-Werror=return-type]
     }
     ^
cc1plus: some warnings being treated as errors
cc1plus: some warnings being treated as errors
```

#### Expected Results:

  * It should have compiled, but due to strict compiler flags, these warnings are treated as errors.

an4cr0n, 2019-06-01:
Yeah we flagged it as experimental because we managed to compile it on our ESP32 but didn't test it well, maybe if you find a solution you can do a pull request, i really coulnd't think why that is especially because it is a newer Arduino version, so it should support latest language features

zrecore, 2019-06-06:
@an4cr0n ...ok, I think I've sorted it out by using static_cast calls where needed, and reworking some if/then logic to make the compiler happy (conditionals with multiple return calls were reworked a bit)

Sonic0, 2019-07-25:
To resolve this problem there is a pull request #145.

zrecore, 2019-08-09:
> To resolve this problem there is a pull request #145.

Sure thing. Here's PR #147

Sonic0, 2019-08-10:
Merged #145

closed 2019-08-10 by Sonic0