F-Army/arduino-dw1000-ng #149: Trilateration

FXUDarkNova, 2019-08-30:
Testing out the StandardRTLSAnchorMain and wondering how I measure the positions of each of my anchor to add to the Position position_self = {0, 0}; so it gets the correction positioning. Also if i attached 3 anchors to a robot and used a tag would trilateration work or would I need to look at alternative method?

Sonic0, 2019-08-30:
Which is the context of your application? 
In standard trilateration application, the anchors are fixed. In this case position_self = {0,0}  is an cartesian coordinate x=0, y=0 (usually for Main anchor).

FXUDarkNova, 2019-08-30:
My idea is all 3 anchors will fixed onto the robot and will then move to wherever the tag is located

an4cr0n, 2019-08-30:
I think you could adopt an alternative approach without using positioning but rather just distance calculation, as the robot will read a reduction in the distance he'll know he is moving in the right direction. Otherwise probably the best approach overall would be to use Phase difference of arrival OR Angle of Arrival algorithms to calculate the position of the tag relative to the robot. Too bad we didn't have time to experiment with PDoA and AoA, but as far as i know there is vast scientific literature about just a google scholar search away. Hope this was of help!

Sonic0, 2019-08-30:
Yes, I think the same! A simple solution is a distance calculator.

FXUDarkNova, 2019-08-31:
Ok thank you for your help do you think would be better to use just two anchors instead of three then?

FXUDarkNova, 2019-09-02:
I have attempted triangulation using two anchors and one tag which seems to be working I just know need to figure out how to do heading any ideas?

Sonic0, 2019-09-04:
Whit two anchors the position of the tag will be inaccurate, but working.
I have no idea at the moment for the algorithm.

FXUDarkNova, 2019-09-05:
Iv managed to get something working using to anchors and one tag work pretty well actually.

closed 2019-09-19 by Sonic0