F-Army/arduino-dw1000-ng #151: Inconsistent ranging

Leylan24, 2019-09-19:
Hi,
So I am trying to get the range between an Anchor and a Tag. I do get a decent range however the ranges change a lot over time. The anchor and tag are at fixed positions however the range between the two keep dropping lower and lower or increasing higher and higherI have plotted the ranges below.

I am looking for any help as to why this is happening. I am using two AA batteries to power the DWM Modules and the grounds and powers are all connected. I have scoped power lines I get a 5% noise tolerance and the power is stable throughout.

I use Single Sided Two Way Ranging as I get better values than the Asymmetric method. I am looking for any help as to why the values are inconsistent. 

[image]

Sonic0, 2019-09-19:
Hi @Leylan24 ,
please give us more information about your hardware, schematics and so on.
Now I can suggest you to not use AA batteries. Voltage supply for DWM1000 is an import thing, in fact, the voltage must be 3,3V. two AA in series battery gives you 3.00V(if you are lucky).

PS: sorry if I  have not replied to you in private

Leylan24, 2019-09-19:
I am using the ESP8266. I've scoped my power lines I get a 3.2V. I will change to a Lipo battery but i think I shall still have the same problem

Leylan24, 2019-09-19:
The reason I think I will have the same problem is because I was using those usb battery packs to power it before switching to the battery

Sonic0, 2019-09-19:
For tests, we use three esp8266 based NodeMCU 1.0 and one Arduino ProMini. With the library's examples and Arduino IDE's integrated compiler, we don't have any problems.
I have no ideas about your problem.

Sonic0, 2019-10-09:
Hi @Leylan24 ,
Do you have news about this issue?

Leylan24, 2019-10-09:
No, cannot seem to figure out why it increases like that.

closed 2019-10-09 by Leylan24