F-Army/arduino-dw1000-ng #152: Smooth Ranging for distance

FXUDarkNova, 2019-09-30:
hi there,

I am using your library file as its more in depth compared to others I have looked however looking at this github https://github.com/thotro/arduino-dw1000  
Is there a way using your library file I can implement a smooth distance filter like this.
//Enable the filter to smooth the distance
  DW1000Ranging.useRangeFilter(true);
I couldn't see anything looking through your code.

Sonic0, 2019-10-01:
Hi @FXUDarkNova,
as soon as possible I will reply to you!

FXUDarkNova, 2019-10-01:
@Sonic0  Thank you

Sonic0, 2019-10-02:
Hi @FXUDarkNova ,
We have removed _DW1000Ranging::useRangeFilter_ because it is a useless function that modifies the range based on the precedent range measure.
We think that this operation must be implemented user side. The user will receive the exact range value and he will handle it based on his necessities.

FXUDarkNova, 2019-10-03:
Hi @Sonic0  

Ok makes sense maybe you could help that I have created a X coord using two sensors but issue i am having is seems really sensitive using it for a robot facing forward ie centre is fine but if I slightly turn I go from 0 to say 120 for example.

Please see code I am using to work out X coord
`x = (pow(dist_left,2) - pow(dist_right,2) + pow(dist_l_r,2)) / (2 * dist_l_r);`

0nism, 2020-08-19:
Was this solved? @FXUDarkNova