F-Army/arduino-dw1000-ng #153: Use the library with STM32F103 ? [Enhancement]

amritsingh87, 2019-10-03:
Do you think it would be possible to use your library with an STM32F103 'Blue Pill' and Arduino ?

Will the code benefit from the 32Bit MCU

I used the guide here 'https://alselectro.wordpress.com/2018/11/18/stm32f103-bluepill-getting-started-with-arduino-core/' to get the F103 programming through the library.

Sonic0, 2019-10-04:
The main problem is the SPI communication, now it is not optimized to other devices than Arduino, esp8266 and esp32.
However, it could work, try it and give us feedback.

asheeshraikwar, 2019-10-05:
Can you please explain in detail about the SPI problems with STM32F103?

Sonic0, 2019-10-05:
Hi @asheeshraikwar,
The DWM1000 ( and all derivated from DW1000 ) has two different required clock for reading and writing inside registers. Some registers (The most important ) required a low SPI clock.
SPI clocks are specified inside DW1000 manual and have maximum values. Actually, these different clocks are static for generic for Arduino and specific for esp8266 and esp32.
It's been months and I don't remember everything. Try it to make sure it doesn't work.

Sonic0, 2019-10-09:
Actually, for devices other than the esp8266 or esp32 the max SPI clock is 16Mhz and min 2Mhz.
The maximum clock supported by DW1000 is 20Mhz.
Give us feedback if you try STM32F103! I'm very interested in this thread.

amritsingh87, 2019-10-10:
Thotro's original library works with stm32 :) 

I will give you an update soon regarding your code on stm

amritsingh87, 2019-10-19:
Apologies for the late update, but this works on F103.

Sonic0, 2019-10-20:
Hi @amritsingh87,
Can you share with the community your schematics or your pin connections with DWM1000?

amritsingh87, 2019-12-03:
Hi!
I used the TREK1000 schematics, but it will only work if you use the unofficial arduino-stm32 library.

amritsingh87, 2019-12-04:
F103-DW1000
PA4-17
PA7-18
PA6-19
AP5-20
PB5-22
PA0-3

Sonic0, 2019-12-06:
Thank you very much! I will use these pin connection to update documentation!!

closed 2019-12-06 by Sonic0