F-Army/arduino-dw1000-ng #154: Anchors send their address if they receive a ping from Tag ?

asheeshraikwar, 2019-10-05:
I want to send a ping message from Tag to the Anchors present in the coverage area and in response, they send their addresses to the Tag, How can I achieve this?

Sonic0, 2019-10-07:
Or you should make your own payload, write this inside buffer registry and send it. To the other side read the buffer, parse payload and send a response.
This is the same operation that you must do with the RTLS standard, but the standard's payloads are recognized by all RTLS devices(that implement it).

asheeshraikwar, 2019-10-19:
Hi, thank you for your response... Anchor Main gets ranges between Anchor B, C, and Tag. I assume two cases here which one is correct or any other possibility.
[image]

Case 1. Is Anchor B and C sending their Range from Tag to Anchor Main?
Case 2. Or Tag sending Ranges from Anchor B and C to Anchor Main?

Sonic0, 2019-10-20:
The right case is the Case 1.

closed 2019-12-03 by asheeshraikwar