F-Army/arduino-dw1000-ng #155: Anchor B,C stop Ranging when Anchor Main is not connected?

asheeshraikwar, 2019-10-19:
Whenever Anchor Main is connected to power Anchor B, C Ranges with Tag but when we disconnect Anchor Main Anchor B, C stop Ranging with Tag. why is that happening?
Can you please provide the Architecture or data flow so we can understand how Anchor Main, B, C are connected to each other and how they ranging with Tag. so I can implement this on more than three Anchors.

Sonic0, 2019-10-20:
This problem occurs because the B and C anchors send the range to the Main anchor. So without the Main anchor nothing works anymore.
This implementation is just an example, it is obvious that then it will have to be modified according to your needs.
To increase the number of anchors, you just declare more anchors and pass them one by one as the next anchor to perform the range. 
In anchor C you have to do as in anchor B: _DW1000NgRTLS::anchorRangeAccept(NextActivity::RANGING_CONFIRM, next_anchor);_ Where "next_anchor" will be the next anchor with which to perform the range.

The _transmitRangeReport()_ function within the examples is a small variation of the ISO/EIC 24730-62 standard. In fact, the standard supposes that every single range is sent to a server that will calculate the position of the tag.

As soon as possible I will include here and in a future documentation the flow standard implemented by us.

asheeshraikwar, 2019-10-21:
Thanks a lot ... really helpful information looking forward for the documentation.

asheeshraikwar, 2019-10-22:
Hi, I am looking for the function which print target Tag address. In the case of multiple Tags, we need to distinguish ranges coming from different Tags.

Sonic0, 2019-10-22:
Hi @asheeshraikwar ,
In this case, you have to define other short addresses to initialize each tag and change the .ino Main example to assign it with _DW1000NgRTLS::transmitRangingInitiation()_ .

closed 2020-08-19 by 0nism