F-Army/arduino-dw1000-ng #157: Increase Speed

FXUDarkNova, 2019-10-28:
Do you have any recommendations on set up to help increase receive distance using 3 anchors and one tag?

Sonic0, 2019-10-29:
Hi @FXUDarkNova ,
by now (with DW**M**1000 as anchors and tag), there are few possibilities:
- Following the specification about the PCB board. See dw1000 user manual.
- Only use vertical polarization.
- Disable "smartPower" and manually set the power at a step up than the default. To perform this, set _false_ the third parameter from up in _device_configuration_t_ struct. So, use _setTXPower()_ API to set manually the transmission power. See dw1000 user manual before trying to do everything about power.

By the way, for every question, there is a response inside Decawave's manuals or application notes.

PS. maybe the title of this issue could change in "Increase receive distance".

an4cr0n, 2019-11-01:
Also be careful to check the current regulatory power limits for your country as you can easily get into illegal range if you don't know what you are doing.

closed 2020-08-19 by 0nism