F-Army/arduino-dw1000-ng #160: Multi tag support

kazmer97, 2019-11-14:
Hi, 
I wanted to ask about the multi-tag support of the system for RTLS. 

I've seen that it was mentioned in a previous question but it was not very elaborate, and the person asking I believe had other setup problems. 

In that thread, the answer was that multi-tag support can be achieved by tweaking the address section in the example code. 
Could you please elaborate on that? 

I am trying to put together a system of 3 or 4 anchors and 3 tags. 
Would the above vaguely mentioned multi-tag support be able to allow deep sleep for the tag? 

Also according to my current understanding of the code, the anchors never go to sleep. Is there an upcoming version where that is planned?

Thanks,

an4cr0n, 2019-11-15:
So first of all every tag must have a different EUI (so they don't receive message intended for others), you can change modifying the following line in the tag example:
https://github.com/F-Army/arduino-dw1000-ng/blob/606b4a234dc7c3800a88f14c42ce9154355f9686/examples/StandardRTLSTag_TWR/StandardRTLSTag_TWR.ino#L83

The simplest way would be to add a pause after every transmission to not have multiple tag talk on top of the other with the anchor. (or better deep sleep, which is indipendent of multi or single tag)

Also regarding the anchors they are not going to sleep because they should always be in alert for blink frames (basically a tag notify) and this is on purpose as they are intended to be connected to main powerline and not to be used with batteries.

We are not planning to release other examples as they are just there to showcase the driver and RTLS abstraction API and they are not meant to be a main component of the library.

What we intend to do (but we find very little time to do this, hope to find it sooner or later) is to release a good documentation that describes the main library so people can reference it without reading the source code directly.

In the end the driver is meant to be used having some prior knowledge about the DWM1000 chip as it is complicated and has a lot of features.

Hope i was of help, let me know !

kazmer97, 2019-11-15:
Hi,

Thank you for the quick reply. This is very helpful.

closed 2019-11-15 by kazmer97

amritsingh87, 2019-12-04:
Multi tag support:
I tried this, and this works, the only issue is that the Anchor is not publishing the Address of the Tag it is ranging with.

How can I get that ?