F-Army/arduino-dw1000-ng #162: Where is Mannual

asheeshraikwar, 2019-12-04:
I am looking for DW1000Ng user manual can you please point me toward it.
[image]

Sonic0, 2019-12-04:
In [WiKi](https://github.com/F-Army/arduino-dw1000-ng/wiki/Decawave's-documentation) there are all links.

asheeshraikwar, 2019-12-04:
Hi thanks for the response, there is any function that gives the tag address from which Anchors ranging?

Sonic0, 2019-12-04:
We are going to answer you as soon as possible to all issues (me or @an4cr0n ).

asheeshraikwar, 2019-12-04:
Thanks, I am really stuck on this.
[image]

an4cr0n, 2019-12-05:
@asheeshraikwar i replied to you in the other issue, i will close this one!

closed 2019-12-05 by an4cr0n