F-Army/arduino-dw1000-ng #163: ESP32 SPI concurrency problem

livius147, 2019-12-12:
**Problem:**
I tried to use the library with the ESP32 dev board: with this configuration the system causes a deadlock with a "interrupt wdt" system error. The interrupt watchdog cannot be refreshed and the ESP32 microcontroller is restarted.

**What is the problem:**
the current SPI implementation of the ESP32 SDK does not allow to call the SPI.usingInterrupt(irq_number) function.
The arduino-dw1000-ng library code in the SPIporting.cpp file is the following:
```
void SPIselect(uint8_t slaveSelectPIN, uint8_t irq) {
                #if !defined(ESP32) && !defined(ESP8266)
                        if(irq != 0xff)
                                SPI.usingInterrupt(digitalPinToInterrupt(irq));
                #endif
                pinMode(slaveSelectPIN, OUTPUT);
                digitalWrite(slaveSelectPIN, HIGH);
        }
```
So, if the ESP32 is defined the usingInterrupt call is not used (it is currently not implemented in the ESP32 SDK).
**BUT** 
the current implementation in the ESP32 causes a deadlock due to the fact that:

- If the main loop accesses the SPI in read and write
- and the interrupt service routine calls the function _readSystemEventStatusRegister() which accesses the SPI port
- then we have a concurrency on the SPI port which causes the deadlock and the WDT problem with the following restart process.

This behavior can be very frequent!!

**Solution:**
The current implementation of the interrupt service routine is -in my opinion- wrong: in general, it is not a good solution to call a long-term/blocking process from the interrupt service routine. So the SPI call from the ISR is deprecated.
The solution is to modify the code this way:

```
volatile intFlag = false;

void interruptServiceRoutine() {
		intFlag = true;
	}

void workqueue() {
		if (!intFlag)
			return;

		intFlag = false;		

		// read current status and handle via callbacks
		_readSystemEventStatusRegister();
		if(_isClockProblem() /* TODO and others */ && _handleError != 0) {
			(*_handleError)();
		}
		if(_isTransmitDone()) {
			_clearTransmitStatus();
			if(_handleSent != nullptr)
				(*_handleSent)();
		}
		if(_isReceiveTimestampAvailable()) {
			_clearReceiveTimestampAvailableStatus();
			if(_handleReceiveTimestampAvailable != nullptr)
				(*_handleReceiveTimestampAvailable)();
		}
		if(_isReceiveFailed()) {
			_clearReceiveFailedStatus();
			forceTRxOff();
			_resetReceiver();
			if(_handleReceiveFailed != nullptr)
				(*_handleReceiveFailed)();
		} else if(_isReceiveTimeout()) {
			_clearReceiveTimeoutStatus();
			forceTRxOff();
			_resetReceiver();
			if(_handleReceiveTimeout != nullptr)
				(*_handleReceiveTimeout)();
		} else if(_isReceiveDone()) {
			_clearReceiveStatus();
			if(_handleReceived != nullptr)
				(*_handleReceived)();
		}
	}

```

We need to call, from the main loop function, the workqueue() function.
So, put inside the main loop the workqueue call:
```
void loop() {
.......
DW1000Ng::workqueue();
........
}
```

**Remarks**
I tried to find a workaround to reproduce in the ESP32 environment the usingRegister() function, but the lack of documentation and the analysis of the SDK source code didn't allowed me to write a similar function. 
In any case the long-term processes should be, in general, put in an external workqueue (as suggested in any OS development manual). So I think the suggested patch could be a reasonable solution to this problem.

an4cr0n, 2019-12-17:
Can you provide a working (at least on your esp32 hardware) pull request? So we can test on our hardware when we have some spare time? Thank you

livius147, 2019-12-19:
I have just tried to push a new branch, but I get the following error:
remote: Permission to F-Army/arduino-dw1000-ng.git denied to livius147.
fatal: unable to access 'https://github.com/F-Army/arduino-dw1000-ng.git/': The requested URL returned error: 403

Sonic0, 2019-12-19:
You have to be a collaborator to be able to push a new branch.
In a few words, this is the procedure to collaborate:
- Fork the target repo to your own account.
- Clone the repo to your local machine.
- Check out a new "topic branch" and make changes.
- Push your topic branch to your fork.
- Use the diff viewer on GitHub to create a pull request via a discussion.
- Make any requested changes.
- The pull request is then merged (usually into the master branch) and the topic branch is deleted from the upstream (target) repo.

Online, There are some guides that can help you to follow these steps.
Thank you so much for your work.

livius147, 2019-12-20:
Done. Thanks for your hints!

closed 2020-08-19 by 0nism