F-Army/arduino-dw1000-ng #165: Example code of StandardRTLSTag_TWR and StandardRTLSAnchor{Main/B/C}_TWR not working

armorsun, 2019-12-26:
### Step 1: What do you want?
* Report an issue.

### Step 2: Describe your environment
 * Arduino IDE version or Toolchain: Arduino 1.8.10 on Mac
 * dw1000ng Library version: Latest
 * Arduino device: Arduino Uno
 * Used adapter board: BB-DW1KM
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * StandardRTLSTag_TWR
  * StandardRTLSAnchorMain_TWR
  * StandardRTLSAnchorC_TWR
  * StandardRTLSAnchorB_TWR

#### Steps to reproduce (for bugs only):

  1. burn the sketches to 4 arduino with dwm1000 modules

#### Observed Results:

  * Nothing happened on serial output of main anchor

#### Expected Results:

  * The expected result is that the Serial of AnchorMain will print the position x and y.

#### Some Experiments:

Experiment 1:
  * I had exam the function of DW1000NgRTLS::receiveFrame() by adding Serial.print("debug") inside it, however, it seems like the Anchor cannot receive the BLINK signal.

Experiment 2:

* By commenting the following code of StandardRTLSTag_TWR:
89    DW1000Ng::applySleepConfiguration(SLEEP_CONFIG);
109    DW1000Ng::deepSleep();
111    DW1000Ng::spiWakeup();
112    DW1000Ng::setEUI("AA:BB:CC:DD:EE:FF:00:00");

The Main Anchor can receive the BLINK signal and then transmit RANGEINIT signal, however, the TAG cannot receive the RANGEINIT signal. The function of TAG, tagTwrLocalize, is returned at fun action waitForNextRangingStep.

Sonic0, 2019-12-27:
Hi @armorsun ,
Arduino Uno has a 5Volt SPI instead of 3,3V required by DW1000. Were you using an SPI adapter board?

armorsun, 2019-12-30:
Yes, the adapter module BB-DW1KM has built in-level shifter.

Sonic0, 2019-12-30:
Well! Online there aren't pictures about the back of this adapter board. I have had a similar problem when the current was inadequate, try to use an external source of power.
unfortunately, we haven't 4 Arduino Uno to test this bug.
Which are your pins connections from Arduino to BB-DW1KM?

armorsun, 2019-12-31:
I will try to use an external source of power as soon as possible.
My pins connections from Arduino to BB-DW1KM is below:
Arduino -> BB-DW1KM
13          -> CLK
12          -> MISO
11           -> MOSI
10          -> SS
9            -> RST
2            -> IRQ
5V         -> VIN
GND      -> GND

armorsun, 2019-12-31:
I had tried using a DC Power Supply as an external source of power, however, the problem still exists.

Sonic0, 2020-01-02:
Actually I have no ideas about your problem. These are hardware we haven't tested.
Note: Interrupt pin (IRQ) is not necessary with those examples because they use initializeNoInterrupt() init function.

armorsun, 2020-01-03:
I will find another module to test it.
Do you have recommendation modules?

armorsun, 2020-01-03:
The code worked now.
I had made two changes:
1. Used an external source of power for TAG, the output of the external source is 9V/1A
2. Commented the three line of codes:
```
109 DW1000Ng::deepSleep();
111 DW1000Ng::spiWakeup();
112 DW1000Ng::setEUI("AA:BB:CC:DD:EE:FF:00:00");
```

I believed that the module consumed more power than Arduino uno can provide when using an USB as the source of power.

Sonic0, 2020-01-07:
1. DWM1000 module consumes about 300 mA during transmission, this is more than Arduino Uno can provide.
2. This is very strange. If you find something let me know.

closed 2020-08-19 by 0nism