F-Army/arduino-dw1000-ng #166: How many tags support working at the same time？

PeterSun01, 2020-01-14:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000ng Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * Filename 1
  * Filename 2

#### Steps to reproduce (for bugs only):

  1. _____
  2. _____
  3. _____

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Sonic0, 2020-01-14:
At the same time, you can perform Two-Way-Ranging with only one Tag.
See [APS013](https://www.decawave.com/wp-content/uploads/2018/10/APS013_The-Implementation-of-Two-Way-Ranging-with-the-DW1000_v2.3.pdf) for more information about TWR.

PeterSun01, 2020-01-15:
> At the same time, you can perform Two-Way-Ranging with only one Tag.
> See [APS013](https://www.decawave.com/wp-content/uploads/2018/10/APS013_The-Implementation-of-Two-Way-Ranging-with-the-DW1000_v2.3.pdf) for more information about TWR.

Think you so much!
I konw 1 slot =1 tag
so  how many slots this demo (StandardRTLSAnchorMain_TWR.ino)support？
For example, there are 2 tags to send blink messages， Does the AnchorMain assign two slots to them？

an4cr0n, 2020-01-19:
No, for the examples we just made devices communicate with each other on a best effort way, there are no real timed slots but rather some hardcoded pause we found empirically when testing on the devices.
Feel free to post a pull request if you manage to write an example with such feature!

closed 2020-01-20 by PeterSun01

jack-vn, 2020-02-11:
hi @Sonic0 , you can support me . I can add more tag (2-10) with this library . 
I'm using Arduino mini + DWM1000