F-Army/arduino-dw1000-ng #167: ESP32 Arduino SPI pin

PeterSun01, 2020-01-19:
### Step 1: What do you want?
* Report an issue.

### Step 2: Describe your environment

  
### Step 3: Describe the problem

18 --> CLK
19 --> MISO
23 --> MOSI
18 --> CS (DECAWAVE)

ESP32 Arduino
pin18 both CLK and CS?

an4cr0n, 2020-01-19:
Didn't notice that, seems like an error to me but maybe @Sonic0  knows better

PeterSun01, 2020-01-20:
@an4cr0n Think you so much!
 @Sonic0 Could you tell me?

Sonic0, 2020-01-20:
Hi @PeterSun01 ,
sorry because of answering now. This pinout has been merged by #147 and is related to specific hardware, the FemtoBeacon ESP32 LPS. I should ask to @zrecore for more clarification.

zrecore, 2020-01-21:
> Hi @PeterSun01 ,
> sorry because of answering now. This pinout has been merged by #147 and is related to specific hardware, the FemtoBeacon ESP32 LPS. I should ask to @zrecore for more clarification.

My apologies, CLK should be GPIO14 (physical pin 17)

zrecore, 2020-01-21:
I'm going to submit a PR that fixes the pins listed in the README... The FemtoBeacon LPS board uses an ESP32-PICO-D4 these days anyway.

PeterSun01, 2020-01-21:
@zrecore  Thinks a lot. so how can I change pins definition. Which file do I modify？

zrecore, 2020-01-21:
> @zrecore Thinks a lot. so how can I change pins definition. Which file do I modify？

I believe that is something that is manually set. See 

https://github.com/F-Army/arduino-dw1000-ng/blob/master/examples/BasicConnectivityTest/BasicConnectivityTest.ino

PeterSun01, 2020-01-21:
> > @zrecore Thinks a lot. so how can I change pins definition. Which file do I modify？
> 
> I believe that is something that is manually set. See
> 
> https://github.com/F-Army/arduino-dw1000-ng/blob/master/examples/BasicConnectivityTest/BasicConnectivityTest.ino

@zrecore  Think you,but I am now using RTLS demo ,StandardRTLSTag_TWR/StandardRTLSAnchorMain_TWR

zrecore, 2020-01-21:
When calling initializeNoInterrupt(), provide ss pin and rst pin parameters
only:

```
void initializeNoInterrupt(uint8_t ss, uint8_t rst) {
initialize(ss, 0xff, rst);
}
```

closed 2020-02-01 by PeterSun01