F-Army/arduino-dw1000-ng #169: hi bro . how to add more tag ??? thanks you

jack-vn, 2020-02-10:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000ng Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * Filename 1
  * Filename 2

#### Steps to reproduce (for bugs only):

  1. _____
  2. _____
  3. _____

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Sonic0, 2020-02-11:
Hi @jack-vn , look at these other issues:
#166 --> How many tags support working at the same time
#160 --> Multi tag support
#161 --> Printing of Tag Address on Anchor Main

Sonic0, 2020-02-13:
Very quickly:

> The simplest way would be to add a pause after every transmission to not have multiple tag talk on top of the other with the anchor. (or better deep sleep, which is indipendent of multi or single tag)

Issue #160 is the most important. Based on the above message from @0nism , You should insert the delay in each TAG, in this way the "Blink" transmissions frame isn't going to occur at the same time.

closed 2020-03-05 by jack-vn