F-Army/arduino-dw1000-ng #170: Arduino M0 Pro

4KARA, 2020-02-13:
I am facing some problems while using the library with Arduino M0 Pro (ATMEL SAMD21G)

 * Arduino IDE version or Toolchain: 1.8.0 (went back to an older version to be sure that the problem is not because of the new updates)
 * dw1000ng Library version: Latest
 * Arduino device: Arduino M0 Pro
  
When upload BasiConnectivityTest.ino, the compiler does not give me any warnings, but the output looks like this:

```
> Device ID: FFFF - model: 255, version: 15, revision: 15
> Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
> Network ID & Device Address: PAN: FFFF, Short Address: FFFF
> Device mode: PAN: FFFF, Short Address: FFFF
> Device ID: FFFF - model: 255, version: 15, revision: 15
> Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
> Network ID & Device Address: PAN: FFFF, Short Address: FFFF
> Device mode: PAN: FFFF, Short Address: FFFF
```

I have read that this issue happens because of bad connections between the module and the micro controller. However, I have checked the connections with multi-meter a few times and I have used the same DWM1000 module with Teensy 3.6 and it works without any issues.

When I upload the BasicSender.ino , the output looks like this :

```
> RX power is [dBm] ... -83.58
> Signal quality is ... 1.00
> Transmitting packet ... #1
> ARDUINO delay sent [ms] ... 0
> Processed packet ... #2
> Transmitting packet ... #2
> ARDUINO delay sent [ms] ... 0
> Processed packet ... #3
> Transmitting packet ... #3
> ARDUINO delay sent [ms] ... 0
> Processed packet ... #4
> Transmitting packet ... #4
> ARDUINO delay sent [ms] ... 0

```
And with BasicReceiver.ino gives output like this:

```
> RX power is [dBm] ... -83.58
> Signal quality is ... 1.00
> Received message ... #1667
> Data is ... ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
> RX power is [dBm] ... -83.58
> Signal quality is ... 1.00
> Received message ... #1668
> Data is ... ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

```

These outputs are printed even without the dwm1000, therefore, I guess that arduino M0 Pro does not even know that dwm1000 is connected.  I do not think that it is hardware problem, because the module works fine with other micro controllers.

Please let me know, if you can solve this problem!!!

Sonic0, 2020-02-13:
This library was not tested for this microcontroller. Your issue is probably related to SPI connection.
First, you should check _SPIporting_ library file to be sure of the compatibility.

4KARA, 2020-02-17:
I was using 11, 12 and 13 I/O pins. But it turned out they are not default SPI pins. I used the ICSP header pins, and it works fine now. Thanks!

Sonic0, 2020-02-17:
Well Done! Can you share with us your pin connections to the dwm1000, in order to update the wiki?

4KARA, 2020-02-18:
Done! I have edited the Readme

Sonic0, 2020-02-19:
> Done! I have edited the Readme

What Readme? There isn't a new pull request

4KARA, 2020-02-19:
> > Done! I have edited the Readme
> 
> What Readme? There isn't a new pull request

Sorry, new to github, don't quite know how things work here. I have made a pull request with my explanation now, I think this should work.

closed 2020-08-19 by 0nism