F-Army/arduino-dw1000-ng #171: Ranging fails from the tag side after elapsed time

kazmer97, 2020-02-13:
### Step 1: What do you want?

When using the Twowayranginginitiator and responder. The ranging fails after 20 seconds to 2 minutes. The elapsed time seems to be almost random but always below the 3 minute mark. 

Any tips of why is this happening? How to avoid it? 

### Step 2: Describe your environment

Both tag and anchor are run from an arduino pro mini 8MHz.

### Step 3: Describe the problem

Based on debug prints the ranging loop gets stuck within the first if for reset inactive. Upon reseting the tag the ranging returns to normal.

Sonic0, 2020-02-16:
Hi @kazmer97 ,
currently, I have no idea about your problem. Have you tried other Arduino examples like _StandardRTLSAnchorMain_TWR.ino_ and _StandardRTLStag_TWR.ino_?

kazmer97, 2020-02-16:
Hi, 

In the meantime I’ve figured it out. The library is trying to lock the SPI speed of any arduino board detected at 16MHz. While the Mini I used could only do 4MHz. The problem ceased to exist once I’ve switched to a Due, which could handle the higher SPI speed.

closed 2020-02-16 by kazmer97

Sonic0, 2020-02-16:
@kazmer97, this is strange. During the developing of this library, we used the Arduino Pro Mini 8MHz 3,3V. I remember that if you try to set a higher SPI clock, Arduino automatically set it maximum clock (in this case 4MHz). 4MHz is enough to read and write from and to the DWM but (minimum is 3MHZ).
Probably this issue is probably introduced by some delay or signal degradation. Nothing can exclude a software bug.
Thank you very much! The resolution of this issue can be very useful to other users.