F-Army/arduino-dw1000-ng #172: Make one unit work as both teg, anchor and msg_rtx...

Frederiknoer, 2020-02-14:
The problem i have is the following:
    - I have 4 modules on the ground and a PC connected to one of them.
    - I would like to know the range in between all unit on the PC
    - This means i need the modules to transmit packages with range, for example ID:0 receives package with range between ID1: and ID:3. 

This does not seem possible now, has anybody done this?
Thanks! :)

Sonic0, 2020-02-16:
Hi @Frederiknoer ,
I think I understand what you would like to obtain. However, an Anchor can't change its role and become a tag.
Your aim is not compliant with RTLS standard (this library is based on it) but you can use the methods of this library, which are specific for each action, to make your own Arduino files.

Bartvelp, 2020-04-29:
@Frederiknoer 
I too am trying to achieve this, did you find a solution?

Frederiknoer, 2020-04-29:
> @Frederiknoer
> I too am trying to achieve this, did you find a solution?

No.. Sadly I did not, I spent a decent amount of hours on with out any results..

Bartvelp, 2020-04-30:
@Sonic0 Could you maybe provide some pointers on which methods would be usable for this?
I would like to develop a system in which I can measure the distance between 2 nodes which are both moving and they both need to know the distance. This does not really fit in the concept of anchors and tags.
Thanks a lot in advance, this library looks very well developed!

0nism, 2020-05-02:
@Bartvelp @Frederiknoer Did you take a look at the examples named TwoWayRanging? I think they could be a good starting point for what you are trying to achieve. The logic is between 2 devices but you can adapt it for multiple of them!
Also feel free to pull request documentation about what you find if it could be useful for others!

CaelanK, 2020-05-28:
@Frederiknoer @Bartvelp Did either of you find a solution to this? I would be interested in any updates! Thanks :)

xSteeljaw, 2020-11-12:
@Bartvelp @Frederiknoer, or anyone else. I'm also interested if someone has found any solution for this situation!