F-Army/arduino-dw1000-ng #173: TWR and Sleep

kazmer97, 2020-02-17:
### Step 1: What do you want?

I would like to ask for help regarding the correct use of sleep implementation during Two Way Ranging. 

### Step 2: Describe your environment

I am running the code on Arduino Dues. The ranging example is set up successfully. 
  
### Step 3: Describe the problem

I am building an application where I would like the tag to go to sleep for a period of time before it ranges again with the anchor. 
Currently, if I run the ranging algorithm continuously on an anchor and I send the tag to sleep, after the first sleep period the range ads a 30-meter error which stays consistent after every subsequent sleep and wakeup of the tag. 

Any ideas on how to avoid this? 

### Code used

```
 * MIT License
 * 
 * Copyright (c) 2018 Michele Biondi, Andrea Salvatori
 * 
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * 
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 * 
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 * SOFTWARE.
*/

/*
 * Copyright (c) 2015 by Thomas Trojer <thomas@trojer.net>
 * Decawave DW1000 library for arduino.
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
 * @file RangingTag.ino
 * Use this to test two-way ranging functionality with two DW1000Ng:: This is
 * the tag component's code which polls for range computation. Addressing and
 * frame filtering is currently done in a custom way, as no MAC features are
 * implemented yet.
 *
 * Complements the "RangingAnchor" example sketch.
 *
 * @todo
 *  - use enum instead of define
 *  - move strings to flash (less RAM consumption)
 */

[... 205 lines cut ...]
            transmitPoll();
            noteActivity();
            // Serial.println("range failed");
        }
    }
}

void loop(){
    

    DW1000Ng::spiWakeup();

    int curr_millis = millis();
    
    while(millis()-curr_millis < 100){
        
        range();
    }
    DW1000Ng::applySleepConfiguration(SLEEP_CONFIG);
    DW1000Ng::deepSleep();
    
    delay(1000);

}
```

kazmer97, 2020-02-17:
Quick Update! 

Setting the antenna delay after wakeup again seems to solve the issue. 

```
 DW1000Ng::spiWakeup();
 DW1000Ng::setAntennaDelay(16436);
```

closed 2020-02-17 by kazmer97

Sonic0, 2020-02-17:
Hi @kazmer97 ,
strange issue. The only function that must be called after _spiWakeup()_ is _setEUI()_. In addition, _applySleepConfiguration()_ should be called one time during the _setup()_.

kazmer97, 2020-02-18:
Hi @Sonic0 ,
Thanks for responding to my posts. 

Regarding the setEUI(), to my understanding that function is necessary for the Extended Unique identifier, based on the DW1000 user manual I don't see how setting that would influence the accuracy of the ranging.

Sonic0, 2020-02-18:
setEUI() function does not influence the accuracy of the ranging.