F-Army/arduino-dw1000-ng #176: Explanation of setTXpower

Bartvelp, 2020-05-12:
### Step 1: What do you want?
* Some help with the setTXpower method

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: 1.8.12
 * dw1000ng Library version: latest (commit cd113ce)
 * Arduino device: ESP8266
 * Used adapter board: DWM1000
  
### Step 3: Describe the problem
I read the DW1000 user manual but I don't understand how to reduce the transmission power, I would like it to be as low as possible.
I saw this https://github.com/F-Army/arduino-dw1000-ng/issues/157. And disabled smart TX power but I don't understand the input I need to give to `DW1000Ng::setTXPower();`, is it an integer that is linear with the output strength? or is it a byte array and what logic is used to construct this byte array?
I would like to have a number that I can tune to my needs. Help would be greatly appreciated.

```c++
device_configuration_t DEFAULT_CONFIG = {
    false, // extendedFrameLength
    true, // receiverAutoReenable
    false, // smartPower
    true, // frameCheck
    false, // nlos
    SFDMode::STANDARD_SFD,
    Channel::CHANNEL_5,
    DataRate::RATE_850KBPS, // RATE_850KBPS RATE_6800KBPS
    PulseFrequency::FREQ_16MHZ,
    PreambleLength::LEN_64,
    PreambleCode::CODE_3
};

interrupt_configuration_t DEFAULT_INTERRUPT_CONFIG = {
    true,
    true,
    true,
    false,
    true
};

void setup_dw1000(){
    Serial.println(F("### DW1000Ng-arduino-node ###"));
    // initialize the driver
    DW1000Ng::initialize(PIN_SS, PIN_IRQ, PIN_RST);
    Serial.println("DW1000Ng initialized ...");
    // general configuration
    DW1000Ng::applyConfiguration(DEFAULT_CONFIG);
    DW1000Ng::applyInterruptConfiguration(DEFAULT_INTERRUPT_CONFIG);

    DW1000Ng::setDeviceAddress(1);
    DW1000Ng::setNetworkId(2);
    
    DW1000Ng::setAntennaDelay(16436);
    DW1000Ng::setTXPower(100);
    
    Serial.println(F("Committed configuration ..."));
    // DEBUG chip info and registers pretty printed
    char msg[128];
    DW1000Ng::getPrintableDeviceIdentifier(msg);
    Serial.print("Device ID: "); Serial.println(msg);
    DW1000Ng::getPrintableExtendedUniqueIdentifier(msg);
    Serial.print("Unique ID: "); Serial.println(msg);
    DW1000Ng::getPrintableNetworkIdAndShortAddress(msg);
    Serial.print("Network ID & Device Address: "); Serial.println(msg);
    DW1000Ng::getPrintableDeviceMode(msg);
    Serial.print("Device mode: "); Serial.println(msg);

    // attach callback for (successfully) sent and received messages
    DW1000Ng::attachSentHandler(handleSent);
    DW1000Ng::attachReceivedHandler(handleReceived);
    delay(10);
    DW1000Ng::startReceive();
}
```

Sonic0, 2020-05-14:
Hi @Bartvelp ,
In this case, you can read the Decawave AP(Application note) [APS023](https://www.decawave.com/wp-content/uploads/2019/07/APS023_Part-1_Transmit_Power_Calibration__Management.pdf) (chapter 4 and 5).
We don't remember everything about this topic. Yes, smart power has to be disabled. There are three different signs for the _setTXPower_ method: 
- byte -> This is the main method that writes the register 0x1E of the DW1000. It takes a byte array of length 4. 
- int32_t -> The integer representation of the previous byte array.
- DriverAmplifierValue & TransmitMixerValue -> This is the simplest method. It takes 2 parameters from 2 enums ( DW1000NgConstants.hpp). As explained in paragraph 5.2 of the APS023, two bytes out of the four in register 0x1E are used and the transmit power in both portions of the message should be the same.

Take a look at the Application Note and in case there are bugs in our code, feel free to send a pull request.

Bartvelp, 2020-05-17:
Thanks a lot! I understand completely now. The DriverAmplifierValue & TransmitMixerValue method really is simple to use. Closing.

closed 2020-05-17 by Bartvelp