F-Army/arduino-dw1000-ng #179: Help Understanding StandardRTLSAnchor Code

johntylerrupert, 2020-08-24:
### Step 1: What do you want?

* Improve code (and need help).
  
### Step 3: Describe the problem
I'm new to GitHub, please let me know if there was a better place to ask this question. I'm working on setting up a system of these DW1000 chips to perform TWR localization. I'm struggling to understand how the anchors initially interact with one another, as I follow the code, I can't find where any of the anchors initiate communication with the others. Any idea what I'm missing here? 

#### Affected file(s) or example(s):
  * StandardRTLSAnchorMain_TWR
  * StandardRTLSAnchorB_TWR
  * StandardRTLSAnchorC_TWR

#### Observed Results:

  * Using these files and three DW1000 chips, the anchors don't seem to communicate with one another. However, the Main anchor will communicate with and locate a tag

#### Expected Results:

  * The anchors communicate with one another. Evidence of this is seeing some Serial output beginning with "Range from: " from the main anchor

johntylerrupert, 2020-08-24:
I dug in a bit deeper and I realize now that Anchors B and C won't send anything to main until they have a distance from the Tag. However, I'm still not getting anything from the main anchor other than the distance to the tag.

Sonic0, 2020-08-25:
Hi @johntylerrupert,
If I have understood well your question, you should get a look at the RTLS ISO/IEC 24730-62 standard. Those examples are based on it. The same for the used methods( _transmitRangingInitiation_, _anchorRangeAccept_ , etc).
Use  _anchorRangeAccept_ to change the Anchor by which the Tag performs the ranging action. The second input parameter in Main and B anchors is the next Anchor (B in Main and C in B). Anchor C does not have a next anchor, then close the loop with _NextActivity:: ACTIVITY_FINISHED_ as first input in _anchorRangeAccept_ method.

johntylerrupert, 2020-08-26:
Yep, that answered my question, thank you.

closed 2020-08-26 by johntylerrupert

Yashvardhan001109, 2022-02-03:
hey @Sonic0 
where can i find this document RTLS ISO/IEC 24730-62 standard, wanted to read it.
Also what other documents do you suggest reading to be able to understand this library and be able to add more features to it.
Thank you

Sonic0, 2022-02-03:
@Yashvardhan001109 the ISO/IEC 24730-62 standard document isn't public and you need to buy it.