F-Army/arduino-dw1000-ng #183: Save antenna delay in EEPROM breaks compatibility with ARM based MCU's

Brezensalzer, 2020-10-19:
Hi,

commit 4309849 breaks this library with all ARM based MCU's as they don't have an EEPROM.
In my case it won't compile for an Adafruit Feather M0 board:

/libraries/arduino-dw1000-ng/src/DW1000Ng.cpp:48:10: fatal error: EEPROM.h: No such file or directory
   48 | #include <EEPROM.h>
      |          ^~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Adafruit Feather M0.

Please make this code (4309849) platform dependant.

Best Regards
Stefan

0nism, 2020-10-20:
Thank you so much for this report and also the pull request, we will review it as soon as possible!

Sonic0, 2020-10-21:
I confirm that changes made by @Brezensalzer works on Arduino's AVR boards like Pro mini and Uno.

Sonic0, 2020-10-25:
PR #184 merged.

closed 2020-10-25 by Sonic0