F-Army/arduino-dw1000-ng #185: Error with Two Way Ranging Initiator and Responder and Help with TWR Example

maeenk, 2020-10-24:
Hello Guys, I am having some trouble when running both the Two Ranging Codes (Initiator and Responder) and also with the 2-D Main Anchor and Tag Implementation codes. I am new to this and want to replicate this system but I've come across some difficulties. Currently, I am using a Node MCU 1.0 ESP 12-e chip that is wired with a DWM1000. Both the Basic Receiver and Sender example codes are working perfectly.  

For the Two Ranging Initiator and Responder. Once I implemented the changes for "Fix ISR #181", the Two Way Responder started to output the following although I believed the changes would help bypass this problem. The Two Way Initiator runs the same edited "Fix ISR #181" changes. 

(Using the example code provided printed a bunch of hex values, and really implementing that changed reduced it. )
[image]

Could you shed some light on the matter? 

Along with this, I am also having difficulty with implementing the Standard RTLS Project. Whenever these codes are uploaded, sometimes the Serial Monitor is blank or it outputs the basic setup string statements but never the ranging. The RTLS project utilizes the original src libraries and not the "Fix ISR #181" changes (that was used only in the Two Way Responder and Initiator).

Tag's Serial Monitor.
[image]

Anchor B and C
[image]

And Anchor Main is usually blank with no range values and once with the printed strings in the setup function. 

Sometimes the output from the tag and anchors' serial monitors may not come out until several minutes or ever (nothing even after 10 minutes). Would you please shed some light to this matter as well. What fixes can I implement?

Thank you in advance!

0nism, 2020-10-25:
Can you try this version? https://github.com/F-Army/arduino-dw1000-ng/releases/tag/0.3

I suggest you proceed in steps, first try to make basic sender and receiver work. Let me know if the stable release works (a lot of code was merged in between and maybe something broke)

maeenk, 2020-10-25:
So I went ahead and tried both the basic sender and receiver tests. These are the outputs I am getting on the Serial Monitor.

Basic Receiver
[image]

Basic Sender
[image]

I am not too sure if I am suppose to see these symbols...

Sonic0, 2020-10-25:
Hi @maeenk,
we have been using the same MCU. Check out the [wiki](https://github.com/F-Army/arduino-dw1000-ng/wiki/Tested-hardware-boards#nodemcu-10-esp8266) to find out pins connection with NodeMCU 1.0, this seems to be an SPI or alimentation problem. Our DWMs use 3.3V out voltage from NodeMCU.
In addition, I had have a problem like this when I make a mistake during the soldering of the DWM1000.

maeenk, 2020-10-25:
You are maybe right. I will double check that. 
[image]

I was wondering if you utilized an external power supply because I am simply using the connection from a usb to a computer. Or a voltage regulator; if so, any recommendations?

Sonic0, 2020-10-25:
> I was wondering if you utilized an external power supply because I am simply using the connection from a usb to a computer. Or a voltage regulator; if so, any recommendations?

We have used the internal voltage regulator of the NodeMCU 1.0 and the Main Anchor connected via computer USB.
Try keeping the WAKE pin to 3.3V. Otherwise, try another dwm1000 device. 
Unfortunatetly, no other ideas.

maeenk, 2020-10-26:
By any chance, were these the connections? These are the physical connections I made. Just want to make sure. Thank you again for your help!

[image]

Sonic0, 2020-10-26:
Yes @maeenk, we have made the same connections but whitout pull-down resistor on IRQ.

maeenk, 2020-11-04:
Hello again, it seemed as if I posted in the wrong post last time. Below includes the conversation from the recent post.

[image]

I wanted to make sure just how exactly do I actually get the same device mode. Does re-downloading the zip file do this? I had switched to an Arduino Uno because I was having difficulty with the NodeMCU. With the Arduino, I was able to get the basic receiver and sender to work. But the two way ranging is still giving me issues. I made sure that I was utilizing 5 volts on the Arduino to use SPI and a bunch of resistors to drop down this voltage to 3.3 volts. However, it doesn't seem to work as I hoped. 
By any chance could I see how the connections were set up or could you give any other advice on the matter? I appreciate all the help too.

Sonic0, 2020-11-04:
Hi @maeenk , check the DEFAULT_CONFIG structs at the beginning of arduino examples.

maeenk, 2020-11-05:
I have made sure that the both serial monitors were printing out the device mode when not using the voltage divider method when testing the two way initiator and responder. Thank you again. I used the two way responder example code; but I do want to point out that in the arduino-dw1000-ng-0.3 version that both the irq and reset pins were commented out, so I uncommented them when running them. I have been following the setup with Arduino Uno below, but used resistors instead of a voltage regulator. 

[image]

Once I used the method with voltage dividers, as Arduino's SPI operates then, I got this from the responder's serial monitor. And this is wrong as I should've seen some range. Maybe it is in something I did but can't seem to find out yet.

[image]

I decided to see what exactly was being sent and received, and this was the responder's serial monitor. This shows that nothing is being sent nor received by the responder.

[image]

Would you provide recommendations? Which zip file should I include instead?

Bartvelp, 2020-11-07:
hey @maeenk 
I have used the ESP8266 and I sort of recognize some of the errors. Please make sure that when the sketch does not print an incorrect Device ID like FFFF in many in your screenshots, or the weird characters printed. These kinds of errors always point to some kind of wiring issue, either I accidentally connected 2 pins together on the DWM (that damn pin spacing is small) or the connection just was not solid enough. Do not use a breadboard, always directly solder a copper wire to both the ESP8266 and to the DWM module, it will save you a lot of headaches. 
Did you do this?
In my tests the dwm1000 isn't too picky about power, just add a 100uF capacitor and you should be fine.
IMO you should stick to the latest version of the library and if you use an ESP8266/ESP32 use the one in my pull request that fixes ISR. But sonic and 0nism know a lot more about this library than I do.
Stick to the basicReceiver and basicSender sketch, when you get that working, continue to ranging sketches.

Sonic0, 2020-11-07:
Yes, when all F appears at the beginning during the reading of configuration committed to the DWM1000, there is a problem with SPI communication. I had the same problems a the beginning of this project due to the breadboards and soldering mistakes w(ﾟДﾟ)w.

maeenk, 2020-11-08:
> In my tests the dwm1000 isn't too picky about power, just add a 100uF capacitor and you should be fine.

Thank you for the response. I too believe that it has something to do with my wiring. I will move to soldering everything instead of breadboarding. I do want to make sure about the power to the dwm1000. Would I connect a 100 uF capacitor from the VCC to VDDAON, VDD3V3 (in the schematic above)? Just want to make sure the connections to delay this any further.

Bartvelp, 2020-11-08:
Add an electrolytic capacitor across VCC (which is the net connected to VDDAON, VDD3V3) and ground. As you might know, the capacitor will act as a (very) small battery that can make sure there is no big voltage drop due to current spikes when transmitting and receiving. Since it buffers a small charge, you always need to connect 1 leg to a positive rail like VCC(/VDD3V3/VDDAON) and 1 leg to ground.
This is my schematic: (BU01 === DWM1000 equivalent)
[image]
The ceramic cap is not really necessary I think but it is common practice.
Hope this helps.

maeenk, 2020-11-12:
Hi guys and thanks for the reply,

I have been troubleshooting the circuit the past four days, just the circuit. I wanted to get some insight from you guys and potentially any solutions. But I decided to take off the dw1000 chip and solely look if my NodeMCU was the problem. Basically, just probing the pins for SPI. My chip select (CS) is working just fine; however, my MOSI, MISO, and clock pins are not giving me anything on the oscilloscope. I do know those signals are essential for the ranging to work. I was hoping maybe I get some insight on what you guys think is happening or possible next step recommendations. Also, where in the libraries those values are set for the NodeMCU? Maybe hardcoding those values into the example codes might fix the issue I have been seeing.

Bartvelp, 2020-11-13:
Hi Maeenk,
Unfortunately I am not too familiar with SPI and the expected voltages, so I can't be of much help in that regard.
But can you further explain what you mean by this: 

> The basic senders and receivers don't utilize SPI, but the Two Way Ranging does.

You cannot connect to the DWM1000 without using SPI, and this library always tries to do so. The default SPI pins are used on the nodemcu, which are D5 D6 and D7, D8 is often used as CS. 
Only CS needs to be defined when using this library without interrupts, see:
https://github.com/F-Army/arduino-dw1000-ng/blob/master/src/DW1000Ng.hpp#L65

Maybe somebody else can help you debug your scope pictures

Sonic0, 2020-11-13:
Hi everyone, sorry Maeenk but I don't remember anything about SPI protocol.

maeenk, 2020-11-18:
By any chance, was there any flashing needed on the decawave chip or the dw1000 chip themselves? If so, do you guys know how to do that or provide me steps on flashing? Can't seem to replicate the examples.

Sonic0, 2020-11-18:
> By any chance, was there any flashing needed on the decawave chip or the dw1000 chip themselves? If so, do you guys know how to do that or provide me steps on flashing? Can't seem to replicate the examples.

Nothing like that, any flashing needed. 
How many DWM1000 do you have tried? Now I remember I broke one DWM1000 (probably during the soldering) and the result was like yours.

benchia001, 2021-01-12:
Try adding a 3 second delay at the setup. It seems to help me with the SPI issues. Perhaps the ESP8266 may not have start-up properly and the SPI setup may not be done properly because of that. (I am not sure of the exact reason but it seems to have helped me)

Also using D5/GPIO15 as a CS seems to cause some boot issues -> https://github.com/esp8266/Arduino/issues/2466. Connect a 4.7k (i saw from other forums that using anything between 1k and 5k should do the trick, I personally used 5k) resistor from GND to D5 which should solve the boot issues. Hope this helps