F-Army/arduino-dw1000-ng #186: Higher Location Rate

CarlosAguilar1986, 2020-10-28:
### Step 1: What do you want?

* Improve code (and need help).
* Successfully used the RTLS codes (3 anchors and 1 tag). I'm getting a new location (x, y) every 200 ms (5 Hz). I would like to know what could I change in the code to have higher rates (maybe 10 Hz or more)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: Arduino IDE 1.8.13
 * dw1000ng Library version: v 0.3
 * Arduino device: Teensy 4.0
 * Used adapter board: ZHONGLIAN  DWM1000  breakout
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):

#### Steps to reproduce (for bugs only):

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

0nism, 2020-10-29:
The main objective of the example is just to show some features and testing. For example anchors exchange data and send results to main anchor which is not a standard use case.

You need to roll out your own rtls code using the api and find the right combination of architecture and delays.As an example you can use delays provided by Arduino and send ranges from anchors to a server in order to reduce air data (UWB) as much as possible, basically the delays become your Hz setting.

CarlosAguilar1986, 2020-10-29:
Thanks for the answer ! This library is really usefull and it works flawlessly with Teensy 4.0 & Teensy 4.1 (at least all the examples). I'll keep exploring until I find the right combination to fit my needs.

0nism, 2020-10-29:
Thank you for your feedback! I appreciated a lot

Sonic0, 2020-10-29:
Hi @CarlosAguilar1986 , Can you share with us your pin connections with Teensy (4 and 4.1)? I would like to improve documentation.

CarlosAguilar1986, 2020-10-29:
Yes of course : 
Connections  : (Teensy 4.0/4.1 -> DWM1000)
 - 3.3V -> VCC
- GND -> GND
- 9 -> RST
- 10 -> CS
- 11 -> MOSI
- 12 -> MISO
- 13 -> CLK
- IRQ -> 2 ( with a 10 kohm resistor to the ground)

CarlosAguilar1986, 2020-10-29:
[image]

CarlosAguilar1986, 2020-10-29:
I am using this breakout for dwm1000 : 
https://fr.aliexpress.com/item/32847501125.html?gps-id=pcStoreJustForYou&scm=1007.23125.137358.0&scm_id=1007.23125.137358.0&scm-url=1007.23125.137358.0&pvid=dddc8fff-7c31-4b90-832c-6a1cc95f0578&spm=a2g0o.store_home.smartJustForYou_828534472.0

Sonic0, 2020-10-29:
Thanks a lot @CarlosAguilar1986

closed 2020-10-29 by Sonic0