F-Army/arduino-dw1000-ng #187: Adruino Pro Compiler Error

d0natz, 2020-11-14:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Suggest a new feature.
* Present a new hardware adapter board or wiring.
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: _____
 * dw1000ng Library version: _____
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * Filename 1
  * Filename 2

#### Steps to reproduce (for bugs only):

  1. _____
  2. _____
  3. _____

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Sonic0, 2020-11-14:
Hi @d0natz, details, screenshots, or anything else?

d0natz, 2020-11-14:
Hi Sonic...

My apologies...
I was not aware that I have pressed the submit button.
I was able to solve the problem by just re-downloading your DW1000-ng library.
Please disregard this issue.
I will try to delete this issue now...

Regards,
Ω

d0natz, 2020-11-14:
Hi Sonic,

It seems that I cannot delete the entry.
Thus, I just uploaded the image of the compiler error -- maybe some people may encounter it in the future.

★SOLUTION★
①DELETE "DW1000-ng" in Arduino IDE library
②DOWNLOAD a fresh "DW1000-ng" library
③ADD to Arduino IDE library

Bartvelp, 2020-11-14:
Hey that's my code, not sure why a redownload would compile.
I just saw that the default arduino EEPROM class does not implement `EEPROM.begin`, only the ESP8266 arduino core.
Should the EEPROM.begin be in an `#ifdef` @Sonic0 ?

d0natz, 2020-11-14:
Hi @Bartvelp,

Just  a brief history...

**2020Oct 30** 
First time I downloaded the "DW1000-ng" library. 
Compilation for ESP8266 & ESP32 was successful, but not for Arduino-ProMini.

**2020Nov14**
I tried to re-download the "DW1000-ng" library, and compilation for Arduino-ProMini has been successful.
I am not sure though, if there were some minor updates with "DW1000-ng" library from Oct 30 ~ Nov 13.

Regards,
Ω

Sonic0, 2020-11-14:
Yes @Bartvelp , we didn't make any test before merge and I had to push a quick fix (look at the master code). Probably @d0natz had the previous version of the code (yours). 
On my NodeMCU 1.0 I have been problem with your code and I made code compitible only with Arduino. Sorry @Bartvelp , I should have talk about it with you, but I have no to much time to spend on this project.

Sonic0, 2020-11-14:
Now, @Bartvelp we could try to re-insert compatility with ESP8266 (and for ESP32?) if you would like to spend time on this feature.

Bartvelp, 2020-11-14:
Ah I see the latest commits. No problem obviously, glad it could be fixed quickly.