F-Army/arduino-dw1000-ng #190: Calculate Position on Tag side

CarlosAguilar1986, 2020-12-17:
### Step 1: What do you want?

* Improve code (and need help).

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: Teensy 4.1
 * dw1000ng Library version: latest
 * Arduino device: _____
 * Used adapter board: Zhonglian brekaout
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  * I al currently working with several Tags and 3 Anchors. I would like to perform the calculations to get the tag location on the tag side (instead of doing it by the main anchor). Do you think it's possible ?

#### Steps to reproduce (for bugs only):

  1. For the moment I have switched several functions (example: DW1000NgRTLS:transmitRangingInitiation is done in the tag side and DW1000NgRTLS::tagTwrLocalize is done in the anchor's side)
  

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

Sonic0, 2020-12-17:
Hi @CarlosAguilar1986, yes it's possible. You have to modify the DW1000NgRTLS files (this implementation is just an example of extraction) to not return the range result to the anchors or make your own workflow taking it as an example.
DW1000NgRTLS files are a simple implementation of the RTLS standard ISO/IEC 24730-62, take a look at it.
There are probably other closed issues with a similar questions.

CarlosAguilar1986, 2020-12-18:
Ok, thanks I will try that :)

Sonic0, 2020-12-18:
I don't remember well about the flow. You should transform anchor main's code like anchor B/C and all Anchors are supposed to send the range to the Tag. The Tag should be like Anchor main. It's harder than I say to you.

CarlosAguilar1986, 2020-12-18:
Ok thanks, yes for the moment I am doing something quite similar. In my case The three anchors send Blinks (instead of having one BLINKI have BLINK_1, BLINK_2 and BLINK_3) and the Tag does the transmitRangingInitiation with each anchor but I have weird results sometimes (maybe some interferences I guess ?).  

Then I read in the official decawave documentation (APS013 APPLICATION NOTE, The implementation of two-way ranging with the DW1000, Page 5 at the bottom) :
"In the case where it is necessary for the tag to be aware of the range, the anchor may optionally send an immediate message to the tag with the calculated TOF"
So I was wondering if your library had this option implemented already since I wanted some better/cleaner solution than what I had.

Sonic0, 2020-12-18:
In our case, the "optional message" would be Range-confirm(Anchor A and B) or activity-finished(Anchor C). In your case, you could modify one of the last steps of the flow (transmitRangingConfirm or transmitActivityFinished) and send back the range in one of two optional final params.
This is an example of a final step payload of an optional message in the RTLS standard:
[image]

Obviously, if you don't want to respect the standard you could refactor the entire DWM1000RTLS class personalizing the payloads. Decawave in their examples does not respect perfectly the standard.

CarlosAguilar1986, 2020-12-19:
Yes, I see now, thanks again for your time :)