F-Army/arduino-dw1000-ng #191: Read ACC_MEM

MaximilianArnold, 2021-01-22:
Hi,

this is not an issue from your side, but a request. I tried to get the CSI from the :
Register file: 0x25 –Accumulator CIRmemory

Register, but every-time  I read the whole register the esp crashes and if I read only partial of it there are only 0s. 

Did you try to read it already?

Sonic0, 2021-01-22:
Hi @MaximilianArnold,
I'll try to help you asap. During this time can you tell me about your ESP model and what function of this library you have used?

MaximilianArnold, 2021-01-23:
Hi,

sure not rush. I use the eps8266 node mcu v3 which works fine. It gives the range and detects the frame correclty.

I could also pass the code which I am using. From the other repo I saw that some register needs to be set:
https://github.com/thotro/arduino-dw1000/issues/83

But I am not sure why I even cannot read the whole register without crashing?

Sonic0, 2021-01-23:
Have you been using one of the functions in this library or your own code?
The _Register file: 0x25 – Accumulator CIR memory_ is a very large bank of data (4Kb), maybe it is necessary to read a few octets step by step. Maybe ESP has not the possibility to store this amount of data in a single variable if this is your strategy (I don't know your code and firmware specification of the ESP family).

MaximilianArnold, 2021-01-23:
actually I tried the debug code as well as my own. 

but 4Kb should be in any dynamic array possible in any esp?

Can you try to read in yours the 0x25? and see if it gets only zeros?

Sonic0, 2021-01-25:
> but 4Kb should be in any dynamic array possible in any esp?
I don't know
> Can you try to read in yours the 0x25? and see if it gets only zeros?
Probably yes, but I don't know when.

MaximilianArnold, 2021-02-05:
Push. I came a little further, but in the end the register is still full of 0s.

MaximilianArnold, 2021-02-09:
You can close it, as there is a racing condition on the wdt and the read of the SPI. Thus, only partial can be read.

closed 2021-02-09 by MaximilianArnold

Sonic0, 2021-02-09:
Thank you very much for this issue, it might be useful for someone.