F-Army/arduino-dw1000-ng #192: Basic Receive test not receiving packets from Sender (HiLetGo ESP-WROOM-32)

MatthewFehl365, 2021-01-22:
### Step 1: What do you want?

* Report an issue.
* Improve code (and need help).
* Present a new hardware adapter board or wiring.

### Step 2: Describe your environment

 * Arduino device:  HiLetGo ESP-WROOM-32
 * Used adapter board: no
  
### Step 3: Describe the problem

Running into issues getting the receive example sketch to work with the ESP32. The sender appears to work correctly and both are setup the same way. Pins were reassigned and the both appear in the basic connectivity test sketch. 

Sender hardware info:
DW1000Ng initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 850 kb/s, PRF: 16 MHz, Preamble: 256 symbols, Channel: #5, Preamble code #3

Receiver hardware info:
DW1000Ng initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 06
Device mode: Data rate: 850 kb/s, PRF: 16 MHz, Preamble: 256 symbols, Channel: #5, Preamble code #3

The basic receiver ESP yields forever after attempting to receive the packet. I'm new to decawave's system and relatively new with these more advanced arduino (and others) projects. need some help trying to find whats going wrong here and if its a larger issue with library compatablility. (or more likely im just dumb and dont know what im doing). 

Thank you for any help! 
Will share code that anyone thinks will help!

Sonic0, 2021-01-22:
Hi @MatthewFehl365,
I'll try to help you asap. During this time, you could try the pull request #178 (Just to keep you busy 😄 ).

MatthewFehl365, 2021-01-25:
Update:

I managed to get the basic receive and send sketches working on the ESP32. I occasionally receive ????? as the message but it looks good for the most part. 

I'm now attempting to use the 2 two ranging examples and no messages are being received or send. It appears that the initiator is never entering:

 // continue on any success confirmation
    if (sentAck) {
        Serial.println("continuing");
        sentAck = false;
        DW1000Ng::startReceive();
    }

Any ideas?

0nism, 2021-01-25:
Hi @MatthewFehl365 , can you try the latest release (not master branch) and see if it works?

0nism, 2021-01-25:
I just realized that the latest release probably didn't work with ESP32, maybe you can try the other pull requests relative to ESP32 and see if they give different results

MatthewFehl365, 2021-01-25:
I am currently using the most recent release. I tinkered around for a bit and managed to get the 2 way ranging examples to run on my esp32's (somewhat). The initiator will send data for some time (approx 10 seconds) and everything works properly but then it pauses and the interrupt watchdog triggers and reboots the esp32.

I'm thinking about modifying it to designate tasks to the 2 cores so it doesn't get hung up on a task. Is this a good approach?

0nism, 2021-01-25:
Oh ok, can you then try the latest master branch? Because there were some pull request merged relative to esp32 maybe they fix this

MatthewFehl365, 2021-01-25:
Ok! I'll update with my progess after I try!

MatthewFehl365, 2021-01-25:
I'm still getting the same error: 

Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008a69e  PS      : 0x00060834  A0      : 0x80089a8f  A1      : 0x3ffbe620  
A2      : 0x3ffb8460  A3      : 0x3ffb8074  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060823  A7      : 0x00000000  A8      : 0x3ffb8074  A9      : 0x3ffb8074  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060821  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40088f2a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008a69e

Backtrace: 0x4008a69e:0x3ffbe620 0x40089a8c:0x3ffbe640 0x4008801b:0x3ffbe660 0x400d368d:0x3ffbe6a0 0x400d301d:0x3ffbe6c0 0x400d2dc8:0x3ffbe6f0 0x400d2e6d:0x3ffbe710 0x400d14f0:0x3ffbe740 0x400d1519:0x3ffbe770 0x400d1f73:0x3ffbe790 0x40080ead:0x3ffbe7b0 0x400845ed:0x3ffbe7d0 0x4008153e:0x3ffb1e20 0x400820b7:0x3ffb1e40 0x40081117:0x3ffb1e60 0x40081157:0x3ffb1e80 0x400d2e53:0x3ffb1ea0 0x400d1570:0x3ffb1ed0 0x400d1f6a:0x3ffb1f00 0x400d10a3:0x3ffb1f20 0x400d11e7:0x3ffb1f40 0x400d1228:0x3ffb1f60 0x400d4a41:0x3ffb1fb0 0x40088215:0x3ffb1fd0

Core 0 register dump:
PC      : 0x40088f35  PS      : 0x00060034  A0      : 0x8008973d  A1      : 0x3ffbe150  
A2      : 0x3ffbee18  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffc0534  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000020  
A14     : 0x00000020  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40088f35:0x3ffbe150 0x4008973a:0x3ffbe180 0x40088303:0x3ffbe1a0 0x4008a515:0x3ffbe1c0 0x400845f6:0x3ffbe1d0 0x400ec73f:0x3ffbbff0 0x400d6e23:0x3ffbc010 0x40089706:0x3ffbc030 0x40088215:0x3ffbc050

Rebooting...

After a short uptime, the wdt timeout is thrown. 

The initiator is getting held up somewhere.

MatthewFehl365, 2021-01-26:
Update:

After some more fiddiling around and searching the other issues, it appears that the power supply is the cause of alot of the problems I've been finding so far. I changed how the modules were powered and that fixed most of my issues. Occasionally the ESP32 wdt error will still be triggered so I'm going to keep troubleshooting but the connection is alot more stable now.

Balabiot, 2021-03-10:
Hi Matthew, i'm doing some tests with esp32 and reading your comments i understand that there can be problems if i don't use good power source. Could you send me your wiring configuration (miso,mosi,cs,irq,rst)? I'm facing some issues with basic receiver/sender and twowayranging examples. Thank you

thire66, 2021-04-30:
Hey @MatthewFehl365 
can you tell me how you managed the two way ranging work? I have nearly the same issue. Connectivity test and and the basic receive and sending examples work. But if i try the two way ranging, the anchor never receive the Poll message.

MatthewFehl365, 2021-04-30:
Make sure you have good clean power going to the module and you are using pull up resistors on the communication lines. Esp32's built in resistors are really iffy. I fixed the problem purly by feeding it clean power

thire66, 2021-05-02:
@MatthewFehl365 
thank you very much for your advice to use pull ups for the SPI lines. It works now.