F-Army/arduino-dw1000-ng #195: 3D position with DW1000 modules

MatthewFehl365, 2021-02-16:
Hello everyone,

Has anyone managed to get 3D positioning working with this library? Also is there any implementation for using many tags and anchors (ie 3-6 tags)? 

Thank you to anyone that as insight!

Sonic0, 2021-02-18:
Hi @MatthewFehl365,
I don't think that someone has made code to add 3D positioning, This library does not implement this feature.

Yashvardhan001109, 2022-02-02:
Hey @MatthewFehl365
Were you able to implement 3d positioning by editing the library, also were you able add more tags and anchors?

MatthewFehl365, 2022-02-02:
Sadly I was not as I have moved onto another project, it should definitely be possible with some version of a multilateration algorithm though. I haven't tried.

Yashvardhan001109, 2022-02-04:
> Hi @MatthewFehl365, I don't think that someone has made code to add 3D positioning, This library does not implement this feature.

Hey
can you explain why it won't be possible to implement 3d positioning using this library, aren't the functions defined enough to get distance from 4 anchors and externally use python or something to implement 3d positioning?

MatthewFehl365, 2022-02-04:
I think you misread my comment, it "should" be possible, this system operates very similarly to GPS so if you were able to implement the same multilateration algorithm implemented for 3d position by GPS it should work here. 

This might be good for some background material on the subject:
https://en.wikipedia.org/wiki/Multilateration 

I haven't read this but the abstract seems interesting:
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6413232/