F-Army/arduino-dw1000-ng #197: analogRead() for ADC1 conflict

elgranshaman, 2021-02-22:
### Step 1: What do you want?

* Report an issue.

### Step 2: Describe your environment
Arduino IDE

 * Arduino IDE version or Toolchain: 1.8.13
 * dw1000ng Library version: v0.3
 * Arduino device: ESP32
 * Used adapter board: custom device
  
### Step 3: Describe the problem
When loading one custom board with the tag example, and another with the anchor example, the anchor board stops the serial output from the two way ranging when the function analogRead() is called. Only happens for ADC1. TWR can start again if the initialization code is repeated.

0nism, 2021-02-24:
Thank you for the report! Can you provide more details?

elgranshaman, 2021-02-25:
Hello,

I am using the code from examples TwoWayRangingInitiator.ino and TwoWayRangingResponder.ino

On the responder example main loop, I have added the line analogRead(36). 

Calling this function which uses peripheral ADC1 stops the TWR function.

If I comment this line, I can see the expected output on the serial monitor of the distance between the two modules. However, if I call this function, there is not serial output anymore.

Only happens for ADC1, if I change to a pin from ADC2, then it works fine.

Sonic0, 2021-02-25:
Hi @elgranshaman, Are you using the latest code from Master branch?

elgranshaman, 2021-02-26:
Hello,

Just downloaded the latest master branch and it's fixed! Thanks a lot and sorry to distrub you

closed 2021-02-26 by Sonic0