F-Army/arduino-dw1000-ng #199: Examples Not Functioning as Expected or Crashing NodeMCU 1.0 (ESP8266)

jrstubbington, 2021-04-18:
### Step 1: What do you want?

* Report an issue.
* Want to report my experience or measurement.

### Step 2: Describe your environment

 * Arduino IDE version: 1.8.13
 * dw1000ng Library version: 0.3 & master
 * Arduino device:  Node MCU 1.0 (ESP8266) (Note: also having similar issues with ESP32-WROOM board)
 * Used adapter board: Custom DWM1000 breakout board (described below)
  
### Step 3: Describe the problem

Using this library, a custom breakout board, and either the NodeMCU 1.0 (ESP8266) or the ESP32 WROOM I am unable to successfully initialize a connection to the DWM1000 module when using `DW1000Ng::initialize(PIN_SS, PIN_IRQ, PIN_RST);`. However, when using `DW1000Ng::initializeNoInterrupt(PIN_SS);` I can successfully run the BasicConnectivityTest and the BasicSender/BasicReceiver (although at very limited range; like 20 feet).

However when attempting the TwoWayRanging examples without the IRQ or RST pins both ESP8266s will print out the connection information for the DWM1000 but never display any other output. So I'm not sure if the IRQ and RST pins are necessary for running more advance examples.

#### Affected file(s) or example(s):
  * BasicConnectivityTest
  * BasicSender / Basic Receiver
  * TwoWayRangingInitiator / TwoWayRangingResponder

#### Steps to reproduce (for bugs only):

  1. Follow pinout instructions from Wiki for ESP8266, including implied connections from examples for RST and IRQ pins
       | Node MCU 1.0 | DWM1000|
       | :----------- | :-----------: |
       | D5 (GPIO14) | CLK |
       | D6 (GPIO12) | MISO |
       | D7 (GPIO13) | MOSI |
       | D8 (GPIO15) | CS | 
       | D2 (GPIO4) | IRQ (5k pull-down resistor) | 
       | D2 (GPIO5) | RST | 
       | 3.3V regulator | 3.3V & VDDAON |
       | GND | GND |
       | X  | PHA (Floating) |
       | X | POL (Floating)  | 

  2.  Using the Arduino IDE, upload any of the example sketches listed above to the ESP8266
  3. _____

Custom breakout board connections for reference: 
[image]

#### Observed Results:

  * What happened?  This could be a description, log output, etc.

When not using the IRQ and RST pins, I can get the following output:
```
DW1000Ng initialized ...
14:38:58.864 -> Committed configuration ...
14:38:58.897 -> Device ID: DECA - model: 1, version: 3, revision: 0
14:38:58.897 -> Unique ID: 00:11:22:33:AA:BB:CC:DD
14:38:58.897 -> Network ID & Device Address: PAN: 0A, Short Address: 05
14:38:58.897 -> Device mode: Data rate: 850 kb/s, PRF: 16 MHz, Preamble: 256 symbols, Channel: #5, Preamble code #3
```

When using **any** example with the IRQ and RST `DW1000Ng::initialize(PIN_SS, PIN_IRQ, PIN_RST);` I receive the following output
```
15:26:47.676 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:26:47.676 -> 
15:26:47.676 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
15:26:47.676 -> 
15:26:47.676 -> load 0x4010f000, len 3584, room 16 
15:26:47.676 -> tail 0
15:26:47.676 -> chksum 0xb0
15:26:47.676 -> csum 0xb0
15:26:47.676 -> v2843a5ac
15:26:47.676 -> ~ld
15:26:47.743 -> ### DW1000Ng-arduino-ranging-tag ###
15:26:47.743 -> ISR not in IRAM!
15:26:47.777 -> 
15:26:47.777 -> User exception (panic/abort/assert)
15:26:47.777 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
15:26:47.777 -> 
15:26:47.777 -> Abort called
15:26:47.777 -> 
15:26:47.777 -> >>>stack>>>
15:26:47.777 -> 
15:26:47.777 -> ctx: cont
15:26:47.777 -> sp: 3ffffe40 end: 3fffffc0 offset: 0000
15:26:47.777 -> 3ffffe40:  00000024 4023c230 00000000 40203688  
15:26:47.777 -> 3ffffe50:  000000fe 00000000 00000000 00000000  
15:26:47.777 -> 3ffffe60:  00000000 00000000 00000000 00ff0000  
15:26:47.777 -> 3ffffe70:  5ffffe00 5ffffe00 676e6967 00000000  
15:26:47.811 -> 3ffffe80:  00000001 00000004 00000005 40204352  
15:26:47.811 -> 3ffffe90:  4010046a 00000000 00000001 40204364  
15:26:47.811 -> 3ffffea0:  009c1001 00003579 00000005 40204879  
15:26:47.811 -> 3ffffeb0:  00000000 3ffee5ec 00000005 00000004  
15:26:47.811 -> 3ffffec0:  3ffe84f6 00000000 3ffee508 00000004  
15:26:47.811 -> 3ffffed0:  3ffe84f6 3ffe84f5 00000005 40204928  
15:26:47.811 -> 3ffffee0:  4020367c 0000000f 00000005 40201d64  
15:26:47.811 -> 3ffffef0:  3fffdad0 00000024 3ffee528 4020392c  
15:26:47.846 -> 3fffff00:  3fffdad0 00000000 3ffee528 3ffee590  
15:26:47.846 -> 3fffff10:  3fffdad0 00000000 3ffee528 402010e8  
15:26:47.846 -> 3fffff20:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.846 -> 3fffff30:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.846 -> 3fffff40:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.846 -> 3fffff50:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.846 -> 3fffff60:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.880 -> 3fffff70:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.880 -> 3fffff80:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.880 -> 3fffff90:  feefeffe feefeffe feefeffe feefeffe  
15:26:47.880 -> 3fffffa0:  feefeffe feefeffe 3ffee550 40203f60  
15:26:47.880 -> 3fffffb0:  feefeffe feefeffe 3ffe8510 40100d65  
15:26:47.880 -> <<<stack<<<
15:26:47.880 -> 
15:26:47.880 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

I have attempted to use both the current _master_ branch as well as the _0.3_ release and both have the same issue.

#### Expected Results:

  * What did you expect to happen?

I would expect the examples to be printing out calculated ranges between the two modules, and eventually use the TWR multi-node positioning examples to calculate a position. Perhaps it is just my inexperience with these modules and SPI in general but I would have also expected the IRQ and RST pin connections to be documented in the Wiki for the ESP8266 if they are actually required.

#### Notes:

I've read in the Issues from the original repository that the DWM1000 can behave strangely if underpowered - I am just supplying the chip from the esp8266s voltage regulator so I've wondered if that's the problem but haven't considered it as related to this particular problem

Sonic0, 2021-04-20:
Hi @jrstubbington,
We have used a very similar [breakout board](https://oshpark.com/shared_projects/95789Glk), so probably ok. The pinout is equal to ours, with a 10k pull-down resistor though.
Did you try other examples like `StandardRTLSTag_TWR.ino` and `StandardRTLSAnchorMain_TWR.ino`?

tisaia, 2021-06-02:
Hello @jrstubbington, aside from @Sonic0 suggestion, perhaps there's also another possibility in the ESP handling of interrupt?
I've tried to do some quick search and found that quite a few people have had trouble with "ISR not in IRAM" issue.
For example: https://forum.arduino.cc/t/using-interrupts-with-a-node-mcu/591995/4

julianblanco, 2021-12-22:
Did you ever get this working? I had similar issues with rcv and send working but ranging did not

Yashvardhan001109, 2022-02-02:
I was getting error in my program when i didn't use ICACHE_RAM-ATTR for interrupts, then i tried using it as shown below. This too didn't work then found that the command must be:
ICACHE_RAM-ATTR void handleInterrupt (){} 
and not void ICACHE_RAM-ATTR

So for this particular library you need to add ICACHE_RAM-ATTR before the void initialize(uint8_t ss, uint8_t irq, uint8_t rst, SPIClass&spi) in DW1000Ng.cpp

Yashvardhan001109, 2022-02-07:
Hey @Sonic0 I have been trying to understand this library, coding is not my strong pursuit. The examples with interrupts are not working for me (I am using noeMCU1.0 and DWM1000). The other examples for anchors and tag is working as it doesn't incorporate interrupts. I wanted to know if you have any draft API for this library as that should make it a lot easier to implement programs. I want to calculate distances between modules with accuracy. Thank you

Sonic0, 2022-02-08:
Hi @Yashvardhan001109, 
I know, examples with interrupts don't work well. You can use the `DW1000Ng::initializeNoInterrupt(<PIN_SS>)` method without problems with NodeMCU1.0. 
I think that programs implementation with DW1000 without minimum know-how isn't a simple goal to reach out to. In this library, there are all methods you need to create medium complex programs and DW1000NgRTLS abstraction (or single methods used) could be a good starting point. Probably, now (after 2 years) I would need a little time to understand what we did.
With the right setup and antenna calibration, you can obtain good accuracy of about 2/3 centimeters.