F-Army/arduino-dw1000-ng #200: Arduino Uno with DWS1000 v0.3 library anchor-tag problem

ImBerkayKaplan, 2021-05-18:
### Step 1: What do you want?
* Report an issue.
* Improve code (and need help).

### Step 2: Describe your environment
 * Arduino IDE version or Toolchain: 1.8.13
 * dw1000ng Library version: 0.3
 * Arduino device: Uno
 * Used adapter board: DWS1000
  
### Step 3: Describe the problem

Although the BasicReceiver and BasicSender are working fine, and my two devices are able to communicate successfully, it seems that the TwoWayRangingResponder - Initiator programs are not working although they seem to be initialized correctly.

#### Steps to reproduce (for bugs only):

  1. Import the arduino-dw100-ng master branch or latest version into Arduino IDE.
  2. Load the examples TwoWayRangingResponder, and initiator into two different Unos that have the DWS1000 shield attached
  3. Open Putty or Serial Monitor to monitor output from both devices

#### Observed Results:

 Although both devices initialize properly, they don't seem to be communicating with each other as there is no other output message other than the configuration details of the devices.
[image]
[image]

#### Expected Results:

I expected to get a distance estimation between the tag and the anchor. If these are not the correct programs, What action should I take to just prepare a simple demo?

Sonic0, 2021-05-21:
Hi @ImBerkayKaplan,
Sorry for the response delay.
In your first screenshot (ranging-anchor) there are some visible problems, fields didn't set up.

tisaia, 2021-06-02:
Hi @ImBerkayKaplan, adding to Sonic comment:
From my experience, one of the possible problem with UNO is its working voltage.
DW1000 works with 3.3V, but UNO GPIO is 5V. 
So, when UNO GPIO sends low signal (0)  to DW1000, DW1000 may still read it as high signal (1). And vice versa.
That makes the connection becomes unreliable and prone to weird results.

closed 2021-10-02 by Sonic0