F-Army/arduino-dw1000-ng #202: Increase range of DWM1000

prisikarm, 2021-09-16:
### Step 1: What do you want?

* Improve code (and need help).
* Want to report my experience or measurement. (Please also add them to the wiki!)

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: latest
 * dw1000ng Library version: latest
 * Arduino device: Teensy 3.2
 * Used adapter board: custom made
  
Hello. 
Currently, I am trying to use DWM1000 to measure distance with as long as possible range. Here are my findings from datasheets on how to improve the range:

`In general, a longer preamble gives improved range performance and better first path time of arrival information while a shorter preamble  gives a shorter air time and saves power.  When operating a low data rate for long range, then a long preamble is needed to achieve that range`

`There are two choices of mean pulse repetition frequency (PRF) within the DW1000.  These are 16 MHz PRF or 64 MHz PRF.  The 16 and 64 are “nominal” since the actual frequencies are related to the 499.2 MHz basic time unit used and differ slightly between preamble and payload parts of the frame.  The higher PRF gives more accuracy on the first path timestamp and perhaps slightly improved operating range, however this comes at the price of additional power consumption. As described above, frames with lower data rate have the largest operating range.`

Based on that recommendations, what do we need to change in library settings in order to achieve the best range possible?

Sonic0, 2021-09-17:
Hi @prisikarm,
What .ino examples currently are you using?
Another important advice, for performance and range accuracy, is the vertical alignment of the antenna. In addition, you could increase the TX power with others +3dB :smirk: (but pay attention, the more TX power the more source power alimentation current needed).

prisikarm, 2021-09-17:
Hi @Sonic0 and thank you for your reply.

> What .ino examples currently are you using?

Currently, I am working with TwoWayRanging. Next step will be to use Anchor/Tag positioning.
My current config is:

`device_configuration_t DEFAULT_CONFIG = {
    false,
    true,
    false,
    true,
    false,
    SFDMode::STANDARD_SFD,
    Channel::CHANNEL_2,
    DataRate::RATE_110KBPS,
    PulseFrequency::FREQ_16MHZ,
    PreambleLength::LEN_2048,
    PreambleCode::CODE_3
};`

> the vertical alignment of the antenna

definitely, that helps a lot (I will add this to my first message)

> In addition, you could increase the TX power with others +3dB 😏

Do you mean to increase +3dB **on top** of what was already introduced in #8?

Sonic0, 2021-09-17:
Hey @prisikarm I'm sorry, I have no time at this moment. I'm going to answer your questions asap.

Sonic0, 2021-09-18:
Hi @prisikarm,
`device_configuration_t` config has those functions under the hood and in your case, you have been setting devices like this:
```
_useExtendedFrameLength() --> False
_setReceiverAutoReenable() --> True
_useSmartPower() --> False
_useFrameCheck() --> True
_setNlosOptimization() --> False
_setSFDMode() --> SFDMode::STANDARD_SFD
_setChannel() --> Channel::CHANNEL_2
_setDataRate() --> DataRate::RATE_110KBPS
_setPulseFrequency() --> PulseFrequency::FREQ_16MHZ
_setPreambleLength() --> PreambleLength::LEN_2048
_setPreambleCode() --> PreambleCode::CODE_3
```
Probably with `setNlosOptimization = True`, `PreambleLength = 1024` and `PulseFrequency = 64MHz` you can obtain a best distance at the expense of range accuracy. Disabling `SmartPower` and setting Power, in manual mode via setTXPower(), can help a lot ( the simplest method sign is `setTXPower(DriverAmplifierValue driver_amplifier, TransmitMixerValue mixer)`.
However, remember that DWM1000 module is very limited due to the integrated antenna. In radio-frequencies works, the best improvement is to install a good antenna.

prisikarm, 2021-09-20:
Thank you for your fast reply @Sonic0. I will try your settings suggestions and I will keep updating the first message. Cheers

prisikarm, 2021-09-24:
Is this is the correct way of adding 3dB for the case when SmartPower switched off?

`byte tx_power_byte = 10100001;`
`DW1000Ng::setTXPower(tx_power_byte);`

where `101` for 2.5dB and `00001` for 0.5db

[image]

Sonic0, 2021-09-24:
> Is this is the correct way of adding 3dB for the case when SmartPower switched off?

Use the third way (the most simple): `DW1000Ng::setTXPower(DriverAmplifierValue::dB_3, TransmitMixerValue::dB_0_5);`
This is simply an abstraction that simplifies the creation of the array.

> `byte tx_power_byte[8] = {1,0,1,0,0,0,0,1};`
> `DW1000Ng::setTXPower(tx_power_byte);`
>
> where `101` for 2.5dB and `00001` for 0.5db

No. The TXPower registry has a 4 byte length, so it would have be something like this: `byte txpower[4] = {'0x0E', '0x08', '0x02', '0x22'};` (this is the default value: [APS023](https://www.decawave.com/wp-content/uploads/2019/07/APS023_Part-1_Transmit_Power_Calibration__Management.pdf)).
However this mode is not used in MANUAL mode, so my advice is not to use this. This is mainly used for calibrations or SMART power mode

You can turn the power up a few more 😉, by the way.

Sonic0, 2021-10-02:
@prisikarm,
can I close this issue?

prisikarm, 2021-10-03:
Yes, thank you so much for your help.

closed 2021-10-03 by prisikarm