F-Army/arduino-dw1000-ng #206: Connecting Arduino Pro Mini with PC

Kabuc0, 2021-12-14:
Hello Together,
I am a complete beginner in this field and hope my question is not too stupid.

I would like to use the dw1000 module with an Arduino Pro Mini. I have already seen the schematic on the wiki and was wondering how to connect the Arduino to the PC. Can I just use a USB to TTL converter to load the code onto the Arduino? And could this also be used as a power supply?

Thanks a lot in advance.

Sonic0, 2021-12-14:
Hi @Kabuc0,

> I am a complete beginner in this field and hope my question is not too stupid.

No problems, all of us were beginners.

> Can I just use a USB to TTL converter to load the code onto the Arduino?

Yes, you can. There are a lot of examples of how to connect USB-TTL to Arduino Pro Mini. However, you need to know what is the version of your Arduino: 3,3V or 5V, and based on the version choose the right voltage from the converter.
This is an example:
[image]

>  And could this also be used as a power supply?

Absolutely not. You can't use the Arduino alimentation. DWM1000s require almost 300 mA and the Arduino does not provide enough current from the current regulator. DWM1000 requires 3,3V as input and I suggest an external power supply for either (remember to not connect Vcc of USB-TTL converter in this case).

**Very important note**: With 5V Arduino you have to use a logic level converter from 5V to 3,3V SPI communication.

Kabuc0, 2021-12-15:
Thanks @Sonic0 for your quick reply. Really appreciate it.

So did I understand that correctly: I have a power connector on a pcb (to which I connect a 3.3V DC power supply) and from that I go to pin 5 and 24 on the DWM1000 and to VCC and GND on the Arduino. And VCC of the USB-TTL Converter is not connected.

And one more thing: Why exactly is there a resistor connecting pin 22 (DWM1000) with ground? Is it because you dont want the module to go in SLEEP/ DEEPSLEEP mode? And its an 330 Ohm Resistor right?

Sonic0, 2021-12-17:
> So did I understand that correctly: I have a power connector on a pcb (to which I connect a 3.3V DC power supply) and from that I go to pin 5 and 24 on the DWM1000 and to VCC and GND on the Arduino. And VCC of the USB-TTL Converter is not connected.

Yes. However, based on the datasheet, I think 3,3V is necessary on pins 5,6,7 and GND on pins 8,16,
21,23,24 (I don't remember well, I used [this board](https://oshpark.com/shared_projects/95789Glk))

> Why exactly is there a resistor connecting pin 22 (DWM1000) with ground? Is it because you dont want the module to go in SLEEP/ DEEPSLEEP mode? And its an 330 Ohm Resistor right?

It is a pull-down resistor. That pin will float in SLEEP and DEEPSLEEP states and may cause spurious interrupts unless pulled low. I remember we used 10k pull-down resistor though.

Kabuc0, 2021-12-17:
okay, thank you really much @Sonic0 !!

closed 2022-02-02 by Sonic0