F-Army/arduino-dw1000-ng #210: No connection: Compatible with Board DWS1000 (Arduino Uno)?

janknspde, 2022-04-14:
* Arduino IDE version or Toolchain: 1.8.19
 * dw1000-ng Library version: 0.3 (downloaded today)
 * Arduino device: Arduino UNO
 * Used adapter board: DWS1000 https://www.mouser.de/datasheet/2/412/DWS1000_Product_Brief_V1_0-1878682.pdf

I have two Arduino UNOs with the shield DWS1000 mounted.
The BasicSender/BasicReceiver is working.

But the TwoWayRangingInitiator/TwoWayRangingResponder is not working.
After the initialization, nothing happens.

``
### DW1000Ng-arduino-ranging-tag ###
DW1000Ng initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: FFFF
Device mode: Data rate: 850 kb/s, PRF: 16 MHz, Preamble: 256 symbols, Channel: #2, Preamble code #3
``

``
### DW1000Ng-arduino-ranging-anchor ###
DW1000Ng initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: 01
Device mode: Data rate: 850 kb/s, PRF: 16 MHz, Preamble: 256 symbols, Channel: #2, Preamble code #3
``

I think the problem is caused by the RST/IRQ pins.
The BasicSender/BasicReceiver examples doesn't need them so it works.

The example code sets the pins RST to 9 and IRQ to 2.
I think for the DWS1000 shield it has to be RST to 7 and IRQ to 8.
But I don't see any effect after changing the pins for both boards.

My problem is similar to these unsolved issues:

- https://forum.qorvo.com/t/how-to-read-dws1000-on-arduino-uno/9428
- https://forum.qorvo.com/t/arduino-uno-with-dws1000-module/10223
- https://forum.qorvo.com/t/arduino-uno-and-dws1000-module-communication-between-two-setups-problem/11392
- https://github.com/F-Army/arduino-dw1000-ng/issues/200

The difference from this issue is, that my boards initialize correct.

I'm very thankful for every hint and idea.
Thank you!

Sonic0, 2022-04-14:
Hi @janknspde,
Has the adapter board SPI voltage converter from 5V to 3,3V? DW1000 works with 3,3V SPI voltage (Arduino Uno has a 5V SPI).
In addition, I suggest you use the latest code from the master branch (version 0.3 is not updated with the latest fixes)

janknspde, 2022-04-19:
Thank you very much for your response. Sorry for the delay, I've been on vacation.
I've used the master branch and not the release tag 0.3, sorry for uncertainty.

I think the shield DWS1000 has logic level converters from 5V to 3V, but I'm not sure.
Would the BasicSender/BasicReceiver work without SPI or wrong SPI voltage??

I've also tested it with an NodeMCU 1.0 as described here: https://github.com/F-Army/arduino-dw1000-ng/wiki/Tested-hardware-boards
I've connected the 4 pins with the DWS1000 shield (SPICLK (R4), SPIMISO, SPIMOSI (R6), SPICSn (R7)) and successful uploaded the BasicConnectivityTest, but it outputs:

````
Device mode: PAN: 00, Short Address: 00
Device ID: 00 - model: 0, version: 0, revision: 0
Unique ID: 00:00:00:00:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: 00
````

I think the shield gets to less power and it needs an external power supply?
The led on the shield lights much darker with the NodeMCU in comparison to the Ardiuno Uno.
What else can I try?

Sonic0, 2022-04-21:
> Would the BasicSender/BasicReceiver work without SPI or wrong SPI voltage??

DW1000 does not work without the current SPI voltage. In the DWS1000 datasheet, I don't read anything about the SPI voltage level converter.

> I think the shield gets to less power and it needs an external power supply?

The NodeMCU 1.0 integrated voltage converter provides enough current to make DW1000 work. Arduino Uno does not provide enough current. In addition, the DWS1000 has an integrated DC/DC converter that can be enabled or disabled via the Jumper J1.

> What else can I try?

I think there is a problem with the SPI connections. In issue #200 the board was the same (DWS1000) and I suppose the problem was the SPI voltage.

janknspde, 2022-04-27:
Hi, I bought some ST Nucleo-F439ZI and use the official DWS1000-Examples from Decawave/Qorvo, and it's working. I had to migrate the examples from F429ZI to work on the F439ZI and had to go deeper in the STM32CubeIDE but I reached the target for now. I have multiple anchors and one single tag which calculates the distances to all anchors.

closed 2022-04-27 by janknspde