F-Army/arduino-dw1000-ng #211: Two-way ranging

L-t-p, 2022-04-28:
### Step 1: What do you want?

* Report an issue.
Hello,When I use the two-way ranging feature, I can't receive the data after a while,Do you have this phenomenon .

* Present a new hardware adapter board or wiring.
This is the schematic I used
[image]

### Step 2: Describe your environment

 * Arduino IDE version or Toolchain: __1.8.14_
 * dw1000ng Library version: ____The latest version_
 * Arduino device: _____
 * Used adapter board: _____
  
### Step 3: Describe the problem

#### Affected file(s) or example(s):
  *  TwoWayRangingInitiator
  * TwoWayRangingResponder  

#### Steps to reproduce (for bugs only):

  1. _____
  2. _____
  3. _____

#### Observed Results:
[image]

  * What happened?  This could be a description, log output, etc.

#### Expected Results:

  * What did you expect to happen?

#### Relevant Code:

```c++
  // TODO(you): code here to reproduce the problem
```

gogojaeuk, 2022-10-24:
hello i have same problem, do you have any solution for this things?

gogojaeuk, 2022-10-24:
@L-t-p