F-Army/arduino-dw1000-ng #212: multiable tags on one device ?

fanno, 2022-05-28:
Hello
I am looking to create a arduino with 3 tags as ilustrated below.

i want it to be "self" calibrating using A1 as x=0 y=0 and i want to leasure distance A1A2 and distance A1 A3 and distance A2A3 Calculate the triangle and create X axis as A1A3 and and create a cartition system automatic from that.

A1=======A2
=
=
=
=
A3

once "calibrated A1-A3 i would be able to add a Tag, and gets it's X/Y positions from that.

However looking at the framework i do not see how this could work with the current source without major rework.
following problems that i have "sporret" may be more:
- "ss" is overwritten (no wat to have more than one at a time)
- Interupt is the same for all and curently no way to determin who the interupt comes from.
      Could make different static classes to be called so you that could determin where it comes from.

only way to make it work that i see is to make dw1000 a none "static" so you could create a DW1000 per device. that would prevent the SS problem. But it dont fix the interupt problem. would have to do something like:

insted of calling "interruptServiceRoutine"

you would call. interruptServiceRoutine1 for IRQ 1 interruptServiceRoutine2 for IRQ 2

then
static uint8_t activeinterupt =0

static DW1000 devices[maxpins]; ???? or something like it. tho it could become a memory problem on smaller devices.

void interruptServiceRoutine1 () {
activeinterupt = 1 < static
interruptServiceRoutine()
activeinterupt = 0
}

void interruptServiceRoutine1 () {
// use the activeinterupt to point to the relevant info needed to know what callbacks ect to reply to. and what SS pin to use ect
devices[activeinterupt]........
}

DW1000NgRanging would propearly also have to be none static

I am fairly new to arduino and c++ so there are good changes i have missed something.

(NOTE: i do not ahve my dw1000 emiters curently so i am curently waiting for that. but wanted to start making the base. of the system)

fanno, 2022-05-29:
Could it be as simple as adding a
void update(uint8_t ss, uint8_t irq, uint8_t rst) {
  _ss = ss;
  _irq = irq;
  _rst = rst;
}

to change the "active" pins depending on and NOT use interupt and "manually"  and:
send out from A1
switch to A2
read from A2 wait for data
Send reply back to A1
switch to A1 
read from A1 wait for data
(and repeat a few times to get a average)

do some thing again for a1/a3 .. and then A3 to a2 ?

once calibrated make loop that loop throw A1-A3 looking for data.

Then reply to T1 when ( A1-A3 got a range command)

A1 would also have to send info about the A1A2. A1A3 and A2A3 so T1 can determin it's position. (this would be done once) per power up

eventuatly if more tags are needed A4 could range to A1-A2-A3 and get their position. of there even more where maybe Aother tag is added around a cornot where it could not see A2 but see A1 and A3 and A4 then it woule use thoes to

noticed from the exambles that the report message has no indentifyer in it.. i assume i could add the "macadress/NetworkId" short or long. to responce so i get to know what range corensponds to what anchor? (adding more data to eg "transmitRangeReport")

mertdluffy, 2022-12-02:
Is there anybody who could manage multiple tag support?